ucb-bar/chipyard #1733: How to make L2 cache write back all the dirty datas when using cache flush instruction？

csgxiong, 2024-01-08:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

I am using the default rocketconfig with L2 cache and c.flush instruction extended.
And when using cache flush instruction，L2 cache did not wrtite back the dirty datas to memory.

### Motivating Example

Are there some configs to L2 cache needed to be configured before using cache flush instruction?
Anyone can help me？
Thanks.

jerryz123, 2024-01-08:
The c.flush instruction only clears the private L1s. To clear the L2s, you can write a physical address to the L2's memory-mapped cache flush control register.

csgxiong, 2024-01-08:
Got it!
Thank you, jerry.

closed 2024-01-08 by csgxiong