ucb-bar/chipyard #1736: Terminal output becomes weird when running build-setup.sh

T-K-233, 2024-01-11:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0
Hash: a6a6a6

### OS Setup

See [here](https://gist.github.com/T-K-233/38e0aff721f2219a96f116822f9dafda)

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

The cursor is moved to the top of the terminal after conda-setup step. I wonder if this is the desired behavior due to some clear screen characters. In previous version the setup script will always print the log appending the existing ones.

### Expected Behavior

N/A

jerryz123, 2024-01-11:
:skull: can you upload a screenshot?

T-K-233, 2024-01-12:
[image]

Note that the cursor is on the top, and the terminal will freeze for ~15 minutes