ucb-bar/chipyard #1740: Error while running Torture test

Sai-Manish, 2024-01-17:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0

### OS Setup

I was trying to run this command in chipyard/sims/verilator path with environment sourced:
make CONFIG=SmallBoomConfig torture

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

I get this following error

`Unrecognized VM option 'MaxPermSize=128M' Error: Could not create the Java Virtual Machine. Error: A fatal exception has occurred. Program will exit. make[1]: *** [Makefile:23: gen] Error 1 make[1]: Leaving directory '/home/saimanish/chipyard_2/tools/torture' make: *** [/home/saimanish/chipyard_2/tools/[torture.mk:12](http://torture.mk:12/): torture] Error 2`

### Expected Behavior

The RISC-V torture utility generates random RISC-V assembly streams, compiles them, runs them on both the Spike functional model and the SW simulator, and verifies identical program behavior.

jerryz123, 2024-01-17:
Try removing the `-XX:MaxPermSize=128M` in torture/Makefile. Please submit a PR to torture if this resolves the problem

Sai-Manish, 2024-01-18:
Hi,
When removed `-XX:MaxPermSize=128M in torture/Makefile` in tools/torture/Makefile, I get the the below following error:
```
Error during sbt execution: java.lang.UnsupportedOperationException: The Security Manager is deprecated and will be removed in a future release
make[1]: *** [Makefile:23: gen] Error 1
make[1]: Leaving directory '/home/saimanish/chipyard_2/tools/torture'
make: *** [/home/saimanish/chipyard_2/tools/torture.mk:12: torture] Error 2
```

jerryz123, 2024-01-18:
In torture/project/build.properties, try updating sbt.version to 1.8.2

Sai-Manish, 2024-01-22:
Thank you Jerry, updating the sbt.version resolves the problem.

closed 2024-01-22 by Sai-Manish

jerryz123, 2024-01-22:
Can you PR the fixes into torture, if you have the time? Thanks

Sai-Manish, 2024-01-23:
Sure will do that.