ucb-bar/chipyard #1771: Update doc for GCD changes

gonsolo, 2024-01-29:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ubuntu

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

See https://github.com/ucb-bar/chipyard/pull/1757
The GCD example was updated to not use TLRegisterRouter but the documentation at
https://chipyard.readthedocs.io/en/latest/Customization/MMIO-Peripherals.html
still refers to it.

### Expected Behavior

Documentation up to date and in line with implementation.

jerryz123, 2024-01-29:
Thanks, I missed this when updating. I'll get to this soon

jerryz123, 2024-01-29:
Actually if you switch to the main branch, you'll see the new docs.

gonsolo, 2024-01-29:
The top of the page still refers to a TLRegisterRouter:

> The easiest way to create a MMIO peripheral is to use the TLRegisterRouter or AXI4RegisterRouter widgets, which abstracts away the details of handling the interconnect protocols and provides a convenient interface for specifying memory-mapped registers.

The GCD changes made me think TLRegisterRouter is obsolete.

PR #1774 (2024-01-30): Update MMIO peripheral docs
Resolves #1771 

**Related PRs / Issues**:

**Type of change**:
- [ ] Bug fix
- [ ] New feature
- [ ] Other enhancement

**Impact**:
- [ ] RTL change
- [ ] Software change (RISC-V software)
- [ ] Build system change
- [ ] Other

**Contributor Checklist**:
- [ ] Did you set `main` as the base branch?
- [ ] Is this PR's title suitable for inclusion in the changelog and have you added a `changelog:<topic>` label?
- [ ] Did you state the type-of-change/impact?
- [ ] Did you delete any extraneous prints/debugging code?
- [ ] Did you mark the PR with a `changelog:` label? [... cut]
closed 2024-01-30 by jerryz123