ucb-bar/chipyard #1776: How can I generate a reset_vector io at DigitalTop module?

csgxiong, 2024-01-31:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

How can I generate a reset_vector io at DigitalTop module?
I want to use the custom reset_vector values to be the boot address. So What should I do?

Thank you!

### Motivating Example

I want to use the custom reset_vector values to be the boot address. So What should I do?

Thank you!

Kevin99214, 2024-02-02:
I had to create this snippet in chipyard.IOBinders and added it to AbstractConfig to be able to pull it out to the DigitalTop level. Hope this helps!

```
class WithExternalResetVectorPunchthrough extends OverrideIOBinder({
  (system: HasTilesModuleImp) => {
    val io_rst_vector_pins_temp = system.reset_vector.zipWithIndex.map { case (dio, i) => IO(dio.cloneType).suggestName(s"rv_$i") }
    (io_rst_vector_pins_temp zip system.reset_vector).map { case (io, sysio) =>
      sysio := io 
    }
    (io_rst_vector_pins_temp, Nil)
  }
})
```

csgxiong, 2024-02-02:
@Kevin99214 
Thank you  for your reply. I really appreciate it.
I had tried your ways, but it does not work. I am using chipyard V1.10 and the  "make verilog CONFIG=RocketConfig" command to generate RTL .
Are there other steps need to be done after  adding  WithExternalResetVectorPunchthrough Config to AbstractConfig?

Kevin99214, 2024-02-02:
Just to make sure but did you set this to true?
`case SubsystemExternalResetVectorKey => true`

csgxiong, 2024-02-02:
It is My mistake.
Now when I set  SubsystemExternalResetVectorKey to be true, the generator shows an exception:
Diplomacy has detected a problem with your graph:
The following node node has 1 inward connections and no outward connections.At least one outward connection was required.
1 inward nodes bound:[once-system.bootROMResetVectorSourceNode]
0 outward nodes bound:[]

Kevin99214, 2024-02-02:
As far as I know, the bootrom needs a reset vector register to program as part of its boot code. If you want an external reset vector, that removes the register the bootrom needs and caused the error. (anyone please correct me if I'm wrong)

For my use case, I rather boot from the address I set using the external reset vector over the bootrom so removing those bootrom configs make sense. But that'll may be different for you. It's up to you to decide if this is what you want.

jerryz123, 2024-02-02:
Setting the `defaultBootAddress` of `BootAddrRegParams` will make the bootrom jump to a different start address.

https://github.com/ucb-bar/testchipip/blob/60a1c15f963222331b8e4127b0ece0ea210786aa/src/main/scala/boot/Configs.scala#L17-L20

csgxiong, 2024-02-04:
@Kevin99214 
Yes. What I want to do is exactly  using the external reset vector to boot the cores.
Thank you for your reply.

csgxiong, 2024-02-04:
@jerryz123 
Thank you, Jerry.
I will gave it a try.

closed 2024-02-04 by csgxiong

csgxiong, 2024-03-05:
@ashutgupta28 
Just like what I had mented before.

When I deleted all the bootrom related configs (withbootrom/withbootaddrReg/withcustombootpin), and set SubsystemExternalResetVectorKey to be true, it finally worked.

ashutgupta28, 2024-03-05:
@csgxiong 
I have not deleted the configs (withbootroom/withbootaddrreg/withcustombootpin) ,
I just comment out it from AbstractConfig and set SubsytemExternalResetVector to true, I am still getting the same error.

Basically , what I did that I made the case class ExternalResetVectorPort in Base Config and there I set SubsytemExternalResetVector => true. 
and then did this.
```
class BaseConfig extends Config(
  new WithDefaultMemPort ++
  new WithDefaultMMIOPort ++
  new WithDefaultSlavePort ++
  new WithTimebase(BigInt(1000000)) ++ // 1 MHz
  new WithDTS("freechips,rocketchip-unknown", Nil) ++
  new WithNExtTopInterrupts(2) ++
  WithExternalResetVectorPort ++
  new BaseSubsystemConfig)
```
but looks like it is working for me.

csgxiong, 2024-03-05:
@ashutgupta28 
Ok, it seems to be a good way.
I will give it a try later.

ashutgupta28, 2024-03-05:
@csgxiong 
Ok. can you tell me the exact changes you have done it for taken the external reset vector out. So that I can give a try. :-)

csgxiong, 2024-03-05:
@ashutgupta28 
My steps are:
1.Delete "case BootROMLocated ***" at BaseSubsytemConfig
2.Delete "WithBootROM" + "WithCustomBootPin " + "WithBootAddrReg"+ "WithCustomBootPinPlusArg" at AbstractConfig 
Then it will work!

ashutgupta28, 2024-03-06:
@csgxiong 
I did the same steps and the RTL is generating but I want to ask you did any top level reset vector port was generating in Digital Top file ?

csgxiong, 2024-03-06:
@ashutgupta28 
Yes.
After doing the steps, there was a reset_vector_0 port at my DigitalTop.

ashutgupta28, 2024-03-06:
Ok. 
Previosly , I have made the new class in rocket-chip/src/main/scala/system/Configs.scala 
```
class WithExternalResetVectorPort extends Config(
  (site, here, up) => {
    case SubsystemExternalResetVectorKey => true
  }
)
```
and then try to invoke it in AbstractConfig . Then I was getting the same error ```
1 inward nodes bound: [once-system.bootROMResetVectorSourceNode]
0 outward nodes bound: []
```
Then I followed your steps like making the ExternalVectorPunchThrough in IOBinders . and removing all the BootRom related config from AbstractConfig . The RTL is generating now but it is not generating any new port in DigitalTop.

Did you also modified the code in BootRom.scala or in other place , because it is not generating in my case,

ashutgupta28, 2024-03-11:
@csgxiong 
Thanks for help.
Now , 
I am able to generate the reset vector out of the DigitalTop as well as Chiptop. But my concern is it is define in BoomFrontEnd.sv 
via this way : 
[image]
[image]

Will this is the same case for you also.. !!

csgxiong, 2024-03-11:
@ashutgupta28 
Nice!!
Well, I find there is a little difference between our work.
I was using the default **RocketConfig**.
It maybe the reasion before.

ashutgupta28, 2024-03-11:
@csgxiong 
So what ever the RTL it is generating in Frontend.v file that worked in RocketConfig ? Right.. In your case
If that worked there then I assumed that it will also worked here also ..

ashutgupta28, 2024-03-13:
@csgxiong 
I have successfully achieved the reset vector and HartID out of the digitalTop Module. And it is fine . 
[image]

But in Chiptop module it is generating as the output and internally driving as zero. 
Any idea how can I change the port as output to input in Chiptop. 
[image]

ashutgupta28, 2024-05-14:
@csgxiong 
Yes.
I have also faced the same problem with this. Because You cannot make the separate Punchthrough for both in IOBinder.
You have to add both the functionality of reset-vector and hart_ID in the same punchThrough otherwise it will override the one.

I have used this snippet

```
class WithExternalHartAndResetVectorPunchthrough extends OverrideIOBinder({
  (system: HasTilesModuleImp) => {
    val ports = system.tile_hartids.zipWithIndex.map { case (eio, i) => {
      IO(eio.cloneType).suggestName(s"hart_id_$i")} }
        (ports zip system.tile_hartids).map { case (io1, sysio1) =>{
        sysio1 := io1 
        }
   }

    val io_rst_vector_pins_temp = system.reset_vector.zipWithIndex.map { case (dio, i) => {
      IO(dio.cloneType).suggestName(s"reset_vector_$i")} }
        (io_rst_vector_pins_temp zip system.reset_vector).map { case (io, sysio) => { 
         sysio := io 
        }
  }
    val mylist = io_rst_vector_pins_temp :++ ports
    (mylist.toSeq, Nil)
  }
})
```

But this will generating the port in DigitalTop nicely. But in Chiptop it is showing the wrong polarity. 
If you know how to do it correctly then let me know.

csgxiong, 2024-05-15:
@ashutgupta28 
Thank you.
That's a good one.
I will give it a try. And if there is anything new, I will keep updating.