ucb-bar/chipyard #1789: GitHub action chipyard-run-tests.yml error

hyun530, 2024-02-07:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0
Hash: b7644b2

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

uname -a : 
Linux SALfpga 5.15.0-67-generic #74~20.04.1-Ubuntu SMP Wed Feb 22 14:52:34 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

lsb_release -a :
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal

printenv :
SHELL=/bin/bash
COLORTERM=truecolor
TERM_PROGRAM_VERSION=1.85.2
CONDA_EXE=/media/0/miniforge3/bin/conda
_CE_M=
VCS_HOME=/media/0/tools/synopsys/vcs/R-2020.12-SP2-11
SBT_HOME=/home/sal/.sdkman/candidates/sbt/current
LC_ADDRESS=ko_KR.UTF-8
JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64
LC_NAME=ko_KR.UTF-8
SDKMAN_CANDIDATES_DIR=/home/sal/.sdkman/candidates
VERDI_HOME=/media/0/tools/synopsys/verdi/R-2020.12-SP2-11
LC_MONETARY=ko_KR.UTF-8
XML_CATALOG_FILES=file:///media/0/miniforge3/etc/xml/catalog file:///etc/xml/catalog
PWD=/media/0
GSETTINGS_SCHEMA_DIR=/media/0/miniforge3/share/glib-2.0/schemas
LOGNAME=sal
XDG_SESSION_TYPE=tty
CONDA_PREFIX=/media/0/miniforge3
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
VSCODE_GIT_ASKPASS_NODE=/home/sal/.vscode-server/bin/8b3775030ed1a69b13e4f4c628c612102e30a681/node
DC_HOME=/media/0/tools/synopsys/syn/Q-2019.12-SP5-5
MOTD_SHOWN=pam
HOME=/home/sal
LANG=en_US.UTF-8
LC_PAPER=ko_KR.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
CONDA_PROMPT_MODIFIER=(base) 
GIT_ASKPASS=/home/sal/.vscode-server/bin/8b3775030ed1a69b13e4f4c628c612102e30a681/extensions/git/dist/askpass.sh
SSH_CONNECTION=115.145.211.249 54403 115.145.211.114 8002
VSCODE_GIT_ASKPASS_EXTRA_ARGS=
LESSCLOSE=/usr/bin/lesspipe %s %s
XDG_SESSION_CLASS=user
SNPSLMD_LICENSE_FILE=27000@semi-lic.skku.edu
TERM=xterm-256color
LC_IDENTIFICATION=ko_KR.UTF-8
_CE_CONDA=
LESSOPEN=| /usr/bin/lesspipe %s
USER=sal
VSCODE_GIT_IPC_HANDLE=/run/user/1000/vscode-git-e66c52e279.sock
CONDA_SHLVL=1
SDKMAN_DIR=/home/sal/.sdkman
SHLVL=2
LC_TELEPHONE=ko_KR.UTF-8
[... 162 lines cut ...]

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

I just want to use chipyard tag 1.10.0 version github action flow in my own server, but it didn't work. I didn't modify anything in all of workflow file. 

[image]
When I ran chipyard-run_tests.yml, problem occured at fireboom-run-tests, prepare-chipyard-cores, prepare-chipyard ~~ case-by-case. 

[image]
All cases the problem was occured at 'Run tests on self-hosted'

[image]
Error log was like this.

[image]
I persumed the problem first occured in this part.

Please let me know what to do or need more information about this issue. 
Thank you. 

### Expected Behavior

I just want to complete the basic workflow

jerryz123, 2024-02-09:
You should check that the chipyard setup flow works manually on your build server first, before attempting to set up the automated CI.

hyun530, 2024-02-14:
What does "setup flow" mean? I believe I have already completed all the setup provided by chipyard. I simply want to add actions to my environment so that the workflow operates when a pull request occurs.

The only modification I made was changing the runner to be self-hosted, and there have been no changes to the content.

Additionally, when accessing the "https://repo.scala-sbt.org/scalasbt/maven-snapshots/org/fusesource/fusesource-pom/1.12/fusesource-pom-1.12.jar" seen in the above download error log, a '404 Not Found' error occurs.
This "404 Not Found" error appears for all links related to the download error.
Is this normal? I suspect there might be an issue with the download link and would like to inquire about this.

Thank you for your response.

jerryz123, 2024-02-14:
I mean you should manually ssh into the self-hosted runner you are attempting to setup, and verify that all the standard chipyard setup steps works in that environment, if you have not done so already. This is what we do when we have issues with our self-hosted runner.

The errors seem to indicate a systemic problem preventing SBT from download packages. Chipyard doesn't do anything special to this process, so I would investigate the run-farm further.

hyun530, 2024-02-16:
I've managed to resolve the previous issue I encountered, thanks for the support.

I have a question regarding the expected outcomes when following the Chipyard tutorials without making any modifications. Specifically, when running the chipyard-run-tests.yml workflow, should all steps succeed without any failures?

I'm asking to ensure that I've set up the dependencies correctly. Any insights or clarifications would be greatly appreciated.

Thank you in advance for your assistance.

jerryz123, 2024-02-16:
All the steps work in our CI, as you can verify from the logs.

hyun530, 2024-02-19:
Thank you I understand.

I'd like to manually check the "prepare chipyard *" flow for each build on the server. Could you please guide me on how to proceed with this process?