ucb-bar/chipyard #1807: make debug fails in verilator

HazemAlindari, 2024-02-28:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.11.0
Hash: ac58f38

### OS Setup

name -a: Linux 8d529d91e146 6.6.12-linuxkit #1 SMP Thu Feb  8 06:36:34 UTC 2024 x86_64 GNU/Linux
[condaList.log](https://github.com/ucb-bar/chipyard/files/14436170/condaList.log)
[printenv.log](https://github.com/ucb-bar/chipyard/files/14436172/printenv.log)

### Other Setup

I started with a fresh chipyard folder and just ruined the build-setup.sh

### Current Behavior

Please excuse my novice knowledge!

I cd to sims/verilator then I build using "make debug > output.log" (same error with "make debug") The build fails with an error:
[terminal.log](https://github.com/ucb-bar/chipyard/files/14436188/terminal.log)
[output.log](https://github.com/ucb-bar/chipyard/files/14436218/output.log)

### Expected Behavior

I expect the build to compile correctly as it compiles problem free with just "make"
So only the debug version is not compiling

abejgonzalez, 2024-02-28:
We recently bumped Verilator due to a Verilator bug. https://github.com/ucb-bar/chipyard/pull/1800. I would check to see if that fixes the issue

HazemAlindari, 2024-02-29:
My bad, I missed this commit entirely, thank you very much, the issue has been fixed!

closed 2024-02-29 by HazemAlindari