ucb-bar/chipyard #1811: 'Debugging with JTAG' section incomplete in docs

sebastianornelas, 2024-03-05:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Include detailed instructions in the Chipyard documentation to debug with JTAG in simulation with OpenOCD + GDB.

### Motivating Example

Need to debug simulation via JTAG in Chipyard. Chipyard links to this page, which leads to nothing. 

See https://chipyard.readthedocs.io/en/stable/Advanced-Concepts/Chip-Communication.html#debugging-with-jtag. Step 3 leads to https://github.com/chipsalliance/rocket-chip#4-launch-openocd.

This leads to the Rocket Chip repo, where the README contains a table of contents with a 'Debugging with GDB' section. This link is broken and/or no documentation exists.

nikhiljha, 2024-03-08:
Here's the fixed link: https://github.com/chipsalliance/rocket-chip/tree/v1.6#4-launch-openocd

jerryz123, 2024-03-08:
A PR to inline these docs into the chipyard docs, since they were removed from rocket-chip, would be greatly appreciated