ucb-bar/chipyard #1812: "Hammer Mentor Plugin" does not exist

T-K-233, 2024-03-06:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.10.0
Hash: [5459aa3](https://github.com/ucb-bar/chipyard/commit/5459aa337126ae8b460166bef79fa2fe1f0770e2)

### OS Setup

[here](https://gist.github.com/T-K-233/38e0aff721f2219a96f116822f9dafda)

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

`vlsi/` is pointing to https://github.com/ucb-bar/hammer-mentor-plugins/tree/eca060af456043ded569bbcc8bea49c03bff6911, which does not exist anymore

### Expected Behavior

NA

T-K-233, 2024-03-06:
Solution: when initializing or updating the repository, do not use `git submodule init` and `git submodule update`

closed 2024-03-06 by T-K-233

jerryz123, 2024-03-06:
That is correct. Either use build-setup, or init-submodules scripts.

waseem-10xe, 2024-08-06:
When I run the **init-submodules-no-riscv-tools** I get following can you confrim.
+ git_submodule_exclude _skip
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip toolchains/riscv-tools/riscv-isa-sim
+ git config --local submodule.toolchains/riscv-tools/riscv-isa-sim.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip toolchains/riscv-tools/riscv-openocd
+ git config --local submodule.toolchains/riscv-tools/riscv-openocd.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip toolchains/riscv-tools/riscv-pk
+ git config --local submodule.toolchains/riscv-tools/riscv-pk.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip toolchains/riscv-tools/riscv-spike-devices
+ git config --local submodule.toolchains/riscv-tools/riscv-spike-devices.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip toolchains/riscv-tools/riscv-tests
+ git config --local submodule.toolchains/riscv-tools/riscv-tests.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip toolchains/riscv-tools/riscv-tools-feedstock
+ git config --local submodule.toolchains/riscv-tools/riscv-tools-feedstock.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip generators/cva6
+ git config --local submodule.generators/cva6.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip generators/nvdla
+ git config --local submodule.generators/nvdla.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip toolchains/libgloss
+ git config --local submodule.toolchains/libgloss.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip generators/gemmini
+ git config --local submodule.generators/gemmini.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip generators/rocket-chip
+ git config --local submodule.generators/rocket-chip.update none
+ for name in toolchains/*-tools/* generators/cva6 generators/nvdla toolchains/libgloss generators/gemmini generators/rocket-chip generators/compress-acc sims/firesim software/nvdla-workload software/coremark software/firemarshal software/spec2017 tools/dsptools tools/rocket-dsp-utils tools/circt vlsi/hammer-mentor-plugins
+ _skip generators/compress-acc
+ git config --local submodule.generators/compress-acc.update none
[... cut ...]
Skipping submodule 'toolchains/riscv-tools/riscv-isa-sim'
Skipping submodule 'toolchains/riscv-tools/riscv-openocd'
Skipping submodule 'toolchains/riscv-tools/riscv-pk'
Skipping submodule 'toolchains/riscv-tools/riscv-spike-devices'
Skipping submodule 'toolchains/riscv-tools/riscv-tests'
Skipping submodule 'toolchains/riscv-tools/riscv-tools-feedstock'
Skipping submodule 'tools/circt'
Skipping submodule 'tools/dsptools'
Skipping submodule 'tools/rocket-dsp-utils'
Skipping submodule 'vlsi/hammer-mentor-plugins'