ucb-bar/chipyard #1849: Updating Doc for Memory Hiearchy page

ksungkeun84, 2024-04-17:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

N/A

### Other Setup

N/A

### Current Behavior

I noticed that the code snippet for cache config in `Memory Hierarchy` page does not exists in the source file anymore.

https://github.com/ucb-bar/chipyard/blob/2f3c83033e86be80bd3410860bb9e5b9d7fbaf89/docs/Customization/Memory-Hierarchy.rst#L21

### Expected Behavior

It would be good to update this page with the latest code for cacne config.

jerryz123, 2024-04-17:
I'm not sure what you mean? That example should still be correct.

ksungkeun84, 2024-04-17:
I'm sorry for your confusion. I actually misunderstoon that code which is an example to configure caches. I thought that code snippet is part of chipyard or rocket chip.

closed 2024-04-17 by ksungkeun84