ucb-bar/chipyard #1887: Torture Test is failing after removing the Floating Point Unit

cygwin24, 2024-05-28:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

I have removed the FPU from the RocketCore. 
This is the configuration I have used to remove it. 

```
class WithNBigCores(
  n: Int,
  overrideIdOffset: Option[Int] = None,
  crossing: RocketCrossingParams = RocketCrossingParams()
) extends Config((site, here, up) => {
  case TilesLocated(InSubsystem) => {
    val prev = up(TilesLocated(InSubsystem), site)
    val idOffset = overrideIdOffset.getOrElse(prev.size)
    val big = RocketTileParams(
    core   = RocketCoreParams(fpu = None),
```

It has been completely removed as I have checked the RTL. 

### Expected Behavior

But the torture test running is fine with FPU Configuration. 

[image]

But it is failing with without FPU Configuration

[image]
But

jerryz123, 2024-05-28:
You likely have to configure torture to not omit FP instructions. You can figure out how to do that through the documentation and source code of torture.

cygwin24, 2024-06-04:
I have entirely removed the FP instruction from torture and also checked that there is no freg register and FP-related instruction generated in the test.S file. But still, my torture test is getting failed