ucb-bar/chipyard #1889: Bare Metal VCU118 capability (similar to Arty100T) and Update to VCU118 documentation

BMorgan1296, 2024-05-29:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Hi,

I have searched the mailing list and found [this](https://groups.google.com/g/chipyard/c/udqI1C0yU8k/m/3gfc74nmDgAJ) which relates to getting the VCU118 to handle bare metal programs loaded onto an SD card.

I would like to know how to change the configs of the VCU118 to support only bare metal programming using the TSI/UART interface just like with the Arty100T. 

I have found that the bootrom for the VCU118 is different to the Arty100T due to the VCU118 config.scala including 'WithSystemModifications -> BootROMLocated' within. However, I have tried swapping out the bootrom here with one from the Arty100T, but I could not send a program over and get any output using the same commands as the Arty. I have also tried commenting out the 'BootROMLocated' section in the hopes that a different one would be loaded for the VCU118. No luck as of yet.

Clearly I have missed something, but I need help as to what. If I could be pointed to the right way to implement this, I would be happy to contribute.

### Update to VCU118 Documentation
The documentation for the VCU118 should also have a notice saying to use the SD card a PMOD to SD card adapter is required. I know that the board specification file says it, but unfortunately I didn't look there first before I went to the shops!

### Motivating Example

I am trying to send binaries over to the Rocket or BOOM CPUs implemented on the FPGA. I have no need for a full Linux boot as I am wanting to compare and benchmark the CPUs running certain tasks. So I would like a way to interface with the VCU118 just like the Arty100T.

jerryz123, 2024-05-29:
Alas, I have not had the time to do this. I would be more than happy to shepherd a PR for this.
I would approach this by creating a new VCU118FPGATestHarness thatcontains only a UART interface for UART-TSI, getting a simple config working with that, then adding back the DDR and GPIO interfaces.

BMorgan1296, 2024-05-29:
I'll give it a go and report back, thanks!

jerryz123, 2024-05-29:
Feel free to open a WIP PR at any point with your work.

BMorgan1296, 2024-05-31:
Got it working but may need some code improvements.