ucb-bar/chipyard #1901: Facing issue while loading bare metal elf file through gdb debugger with openocd + gdb  over virtual Jtag setup on actual fpga emulator on RISCV soc.

tushar1277, 2024-06-13:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: [1.11.0]
Hash: [ac58f38]

### OS Setup

Fpga emulator : Federa linux OS with kernel 4.18.0 and arch of type x86_64.
chipyard repo is build from release 1.11.0 on Ubuntu 22.04 , 
Generated rtl is being used design custom riscv based soc which is running on actual fpga emulator.
For debugging we have virtual jtag + openocd +gdb setup to debug the soc on fpga emulator

### Other Setup

NA

### Current Behavior

From the logs we can see the connection between virtual jtag +  openocd and gdb is done successfully. but when the moment I try to load the .elf file using gdb using load command then I am seeing lot of timeout issues on openocd setup.  attaching the generated rtl dts file in chipyard repo.

[chipyard.harness.TestHarness.RocketConfig.txt](https://github.com/user-attachments/files/15815262/chipyard.harness.TestHarness.RocketConfig.txt)

Let us know what could be the possible root cause of the  issue here ?

**GDB logs** 
(gdb) tar ext:3333
Remote debugging using :3333
0x0000000000010040 in ?? ()
(gdb) monitor targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* riscv.cpu          riscv      little riscv.cpu          halted

(gdb) load
Loading section .text.init, size 0x204 lma 0x80000000
Loading section .tohost, size 0x48 lma 0x80001000
Loading section .text, size 0x3c4c lma 0x80001048
Ignoring packet error, continuing...
Loading section .text.startup, size 0x14 lma 0x80004c94
Load failed
(gdb) 

**openocd logs**

Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, misa=0x800000000094112d
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* riscv.cpu          riscv      little riscv.cpu          halted

Error: Timed out after 2s waiting for busy to go low (abstractcs=0x10001008). Increase the timeout with riscv set_command_timeout_sec.
Error: Abstract command ended in error 'busy' (abstractcs=0x10001108)
Error: Timed out after 2s waiting for busy to go low (abstractcs=0x10001108). Increase the timeout with riscv set_command_timeout_sec.
Error: Abstract command ended in error 'busy' (abstractcs=0x10001108)
Error: Timed out after 2s waiting for busy to go low (abstractcs=0x10001108). Increase the timeout with riscv set_command_timeout_sec.
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (8222 ms). Workaround: increase "set remotetimeout" in GDB
Error: Abstract command ended in error 'busy' (abstractcs=0x10001108)
Error: Timed out after 2s waiting for busy to go low (abstractcs=0x10001108). Increase the timeout with riscv set_command_timeout_sec.
Error: Failed to read priv register.
Error: Abstract command ended in error 'busy' (abstractcs=0x10001108)
Error: Timed out after 2s waiting for busy to go low (abstractcs=0x10001108). Increase the timeout with riscv set_command_timeout_sec.
Warn : negative acknowledgment, but no packet pending
Warn : negative acknowledgment, but no packet pending
Warn : negative acknowledgment, but no packet pending
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet n
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (8222 ms). Workaround: increase "set remotetimeout" in GDB
Error: Abstract command ended in error 'busy' (abstractcs=0x10001108)
Error: Timed out after 2s waiting for busy to go low (abstractcs=0x10001108). Increase the timeout with riscv set_command_timeout_sec.
Error: Failed to read priv register.
Error: Abstract command ended in error 'busy' (abstractcs=0x10001108)
Error: Timed out after 2s waiting for busy to go low (abstractcs=0x10001108). Increase the timeout with riscv set_command_timeout_sec.
Warn : negative acknowledgment, but no packet pending
Warn : negative acknowledgment, but no packet pending
Warn : negative acknowledgment, but no packet pending
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (5999 ms). Workaround: increase "set remotetimeout" in GDB
Error: Abstract command ended in error 'busy' (abstractcs=0x10001108)
Error: Timed out after 2s waiting for busy to go low (abstractcs=0x10001108). Increase the timeout with riscv set_command_timeout_sec.
Error: Failed to read priv register.
Error: Abstract command ended in error 'busy' (abstractcs=0x10001108)
Error: Timed out after 2s waiting for busy to go low (abstractcs=0x10001108). Increase the timeout with riscv set_command_timeout_sec.
Warn : negative acknowledgment, but no packet pending
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (7006 ms). Workaround: increase "set remotetimeout" in GDB
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet n

### Expected Behavior

elf file loading at memory (0x80000000) through gdb  should be done successfully without any error/warning from openocd.

tushar1277, 2024-06-19:
The issue was inside our custom generated rtl hence closing this bug

frideric, 2025-05-06:
We encountered the same problem as you, with the same symptoms and errors. What is the specific problem in RTL you mentioned? How to locate and debug the problem?