ucb-bar/chipyard #1915: Is the data in the inclusive L2 cache synchronized with the data in the DDR, or how can I enable the write-through function of the L2 cache?

Yuxin-Yu, 2024-07-03:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

no

### Motivating Example

no

jerryz123, 2024-07-03:
There is no write-through capability currently implemented.

For write-through support, you can write to the MBUS bypass mechanism if you enable it, note that this will be an incoherent write to backing memory.

Yuxin-Yu, 2024-07-04:
Hi @jerryz123 ,Thanks for your reply. Do you have any plans to add write-through functionality? Also, could you be more specific about the MBUS bypass mechanism you mentioned?

jerryz123, 2024-07-17:
I don't have plans to add write-through functionality.
Look at `SbusBypassRocketConfig` and `WithExtMemSbusBypass`. These configs add an address alias (OR in a high-bit into your physical address), through which the system will bypass the L2 and incoherently access DRAM.

Yuxin-Yu, 2024-07-18:
Thank you @jerryz123 , I'll give it a try

closed 2024-07-18 by Yuxin-Yu

Yuxin-Yu, 2024-07-19:
Hi @jerryz123 ,I found that when using the default Banked L2, direct access to DRAM can be achieved through WithExtMemSbusBypass configuration. However, when using Inclusive L2, the WithExtMemSbusBypass configuration does not work. Is this normal?