ucb-bar/chipyard #1922: Chipyard1.110 version, vcs make failed

YeyeQian, 2024-07-18:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.11.0

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`
Linux server146 3.10.0-862.el7.x86_64 #1 SMP Wed Mar 21 18:14:51 EDT 2018 x86_64 GNU/Linux

### Other Setup

VCS version: N-2017.12-1

### Current Behavior

I tried to run the vcs simulation in chipyard1.11.0, but it failed with the following informatiion:

g++ -w  -pipe -fPIC -std=c++17 -I/data/userhome/zhaoyifan/Work/chipyard/.conda-env/riscv-tools/include -I/data/userhome/zhaoyifan/Work/chipyard/tools/DRAMSim2 -I/data/userhome/zhaoyifan/Work/chipyard/sims/vcs/generated-src/chipyard.harness.TestHarness.LoopbackNICRocketConfig/gen-collateral -O3 -I/eda/apps/snps/vcs-mx/N-2017.12-1/include    -c /data/userhome/zhaoyifan/Work/chipyard/sims/vcs/generated-src/chipyard.harness.TestHarness.LoopbackNICRocketConfig/gen-collateral/uart.cc
cc1: warning: command-line option '-std=c++17' is valid for C++/ObjC++ but not for C
/data/userhome/zhaoyifan/Work/chipyard/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/11.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: /eda/apps/snps/vcs-mx/N-2017.12-1/linux64/lib/vcs_save_restore_new.o: relocation R_X86_64_32S against symbol `_sigintr' can not be used when making a PIE object; recompile with -fPIE
/data/userhome/zhaoyifan/Work/chipyard/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/11.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:107: product_timestamp] Error 1
make[1]: Leaving directory '/data/userhome/zhaoyifan/Work/chipyard/sims/vcs/generated-src/chipyard.harness.TestHarness.LoopbackNICRocketConfig/chipyard.harness.TestHarness.LoopbackNICRocketConfig'
Make exited with status 2
make: *** [Makefile:72: /data/userhome/zhaoyifan/Work/chipyard/sims/vcs/simv-chipyard.harness-LoopbackNICRocketConfig] Error 2

### Expected Behavior

I have searched the solutions, including this one https://github.com/ucb-bar/chipyard/issues/1011, but none of them works. 
I suspect there is a mismatch between the vcs version and the gcc version, but I don't know how to change the gcc version in the conda environment.

jerryz123, 2024-07-19:
This is an unfortunate problem that arises when the libraries installed by the conda environment conflict with the system libraries expected by VCS.

Unfortunately, the only solution for this at the moment is to avoid using the conda environment entirely, and install the riscv-gnu-toolchain, riscv-isa-sim, circt, and other tools manually.

You can run `build-setup.sh` with the right set of flags to omit generating a Conda environment, in which case you must provide your own `$RISCV` directory into which the tools will be installed. After installing the tools into `$RISCV`, adding `$RISCV/bin` to your PATH should be sufficient to get the repo working with your system VCS.

Keep in mind when going down this path that you will have to manage and resolve any missing system dependencies required by various tools in Chipyard's dependencies. This can be a tedious process.

YeyeQian, 2024-07-23:
Thank you very much. I'll give it a try