ucb-bar/chipyard #1934: Tutorial for sky130 vlsi fails due to unbounded memory usage

TheMichaelGionet, 2024-07-30:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

commit d0a9520

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`
uname -a: 
Linux Beef 6.5.0-41-generic #41~22.04.2-Ubuntu SMP PREEMPT_DYNAMIC Mon Jun  3 11:32:55 UTC 2 x86_64 GNU/Linux

lsb_release -a: 
No LSB modules are available.
Distributor ID:   Ubuntu
Description:      Ubuntu 22.04.4 LTS
Release:    22.04
Codename:   jammy

printenv:
SHELL=/bin/bash
SESSION_MANAGER=local/Beef:@/tmp/.ICE-unix/1113,unix/Beef:/tmp/.ICE-unix/1113
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
SSH_AGENT_LAUNCHER=gnome-keyring
XDG_MENU_PREFIX=gnome-
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
CONDA_EXE=/home/michael/miniconda3/bin/conda
_CE_M=
LANGUAGE=en_CA:en
JAVA_HOME=/home/michael/Documents/chipyard/.conda-env/lib/jvm
GNOME_SHELL_SESSION_MODE=ubuntu
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
JAVA_LD_LIBRARY_PATH=/home/michael/Documents/chipyard/.conda-env/lib/jvm/lib/server
XMODIFIERS=@im=ibus
DESKTOP_SESSION=ubuntu
XML_CATALOG_FILES=file:///home/michael/Documents/chipyard/.conda-env/etc/xml/catalog file:///etc/xml/catalog
GTK_MODULES=gail:atk-bridge
DBUS_STARTER_BUS_TYPE=session
PWD=/home/michael/Documents/chipyard/sims/verilator
GSETTINGS_SCHEMA_DIR=/home/michael/Documents/chipyard/.conda-env/share/glib-2.0/schemas
LOGNAME=michael
XDG_SESSION_DESKTOP=ubuntu
XDG_SESSION_TYPE=wayland
CONDA_PREFIX=/home/michael/Documents/chipyard/.conda-env
SYSTEMD_EXEC_PID=1113
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
XAUTHORITY=/run/user/1000/.mutter-Xwaylandauth.JXUVQ2
HOME=/home/michael
USERNAME=michael
IM_CONFIG_PHASE=1
LANG=en_CA.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=ubuntu:GNOME
VTE_VERSION=6800
WAYLAND_DISPLAY=wayland-0
CONDA_PROMPT_MODIFIER=(/home/michael/Documents/chipyard/.conda-env) 
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/dbecb6cf_c238_4a6a_848f_238fa92b613d
GNOME_SETUP_DISPLAY=:1
LESSCLOSE=/usr/bin/lesspipe %s %s
XDG_SESSION_CLASS=user
TERM=xterm-256color
_CE_CONDA=
LESSOPEN=| /usr/bin/lesspipe %s
USER=michael
[... 444 lines cut ...]
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xxhash                    0.8.0                h7f98852_3    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.9.4           py310h2372a71_0    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstandard                 0.22.0          py310h1275a96_0    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...
After a fresh setup with that commit, follow https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html

Then https://chipyard.readthedocs.io/en/latest/VLSI/Sky130-OpenROAD-Tutorial.html

### Current Behavior

When pulling a fresh repo, doing the setup, then following the steps in this tutorial https://chipyard.readthedocs.io/en/latest/VLSI/Sky130-OpenROAD-Tutorial.html, I am getting an issue where during the placement stage, the amount of memory that is used increases without bound. All 200 GB of swap memory in my Ubuntu setup gets consumed, then ubuntu kills openroad. 

### Expected Behavior

Memory will stay within a reasonable range during the place stage.

jerryz123, 2024-07-30:
This is likely an issue with OpenROAD, you should see if they have any advice

Jerry-Tianchen, 2024-08-26:
I solve the issue by compiling the newest Openroad with GUI. However, more places are broken:

- SRAM Macro info is not correct. TinyRocketConfig only has 1 DCache Data array, 1 ICache Data Array, and 1 ICache Tag Array, but the **_yml_** file seems to contain the wrong number of DCache SRAM Macros.
It seems Syn tools don't support the Async Reset FF found in the TinyRocketConfig. I don't know how they successfully generates the chip in the pervious chipyard version, perhaps at that time we don't have async reset in the chipyard?