ucb-bar/chipyard #1975: Error while executing the command " ./build-setup.sh "

waseem-10xe, 2024-08-07:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

When I execute the ./build-setup.sh command i get this:
[image]

### Expected Behavior

 Built successfully.

JFKTRSDAKIDHIKA, 2024-08-09:
i also run in to thuis problem.
the way i address this is that i rrmove the chipyard fold and clone a new one and ckeckout to branch HEAD instead of branch 1.4

waseem-10xe, 2024-08-09:
Yes I also have solved the error by running the commands ./build-setup.sh riscv-tools -s 9
If I skip the step 9 then build setup is done an d work properly.

closed 2024-08-09 by waseem-10xe