ucb-bar/chipyard #1986: Can not boot linux on Shutle

franktaTian, 2024-08-09:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

main 

### OS Setup

Linux i7700 5.4.0-189-generic #209-Ubuntu SMP Fri Jun 7 14:05:13 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

I Can boot Linux (Built by Firemarshal ) on my local VCU118  board.But when I try to boot the same image with Shuttle CONFIG, it hung at  somewhere. 

### Expected Behavior

Boot the same linux kernal correctly as RocketChip CONFIG.

### Other Information

OpenSBI v0.8-3-g508f325
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : ucb-bar,chipyard
Platform Features   : timer,mfdeleg
Platform HART Count : 1
Boot HART ID        : 0
Boot HART ISA       : rv64imafdcsu
BOOT HART Features  : scounteren,mcounteren
BOOT HART PMP Count : 0
Firmware Base       : 0x80400000
Firmware Size       : 88 KB
Runtime SBI Version : 0.2

MIDELEG : 0x0000000000000222
MEDELEG : 0x000000000000b109
[    0.000000] OF: fdt: Ignoring memory block 0x8000000 - 0x8010000
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80600000
[    0.000000] Forcing kernel command line to: console=hvc0 earlycon=sbi
[    0.000000] Linux version 5.7.0-rc3-58541-g5480facf4315-dirty (peter@i7700) (gcc version 9.2.0 (GCC), GNU ld (GNU Binutils) 2.32) #49 SMP Fri Aug 9 15:04:28 CST 2024
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] initrd not found or empty - disabling initrd

jerryz123, 2024-08-09:
Try checking out this version of shuttle. 7e2afd188f84b349eb1843e11ff0d8168b22f8a7. 
I'll update its pointer in chipyard soon

franktaTian, 2024-08-10:
> Try checking out this version of shuttle. 7e2afd188f84b349eb1843e11ff0d8168b22f8a7. I'll update its pointer in chipyard soon

Still cannot resolve the problem yet. Do you have any workable  Shuttle  and firemarshal configurations on firefim?

franktaTian, 2024-08-10:
> Try checking out this version of shuttle. 7e2afd188f84b349eb1843e11ff0d8168b22f8a7. I'll update its pointer in chipyard soon

Yes!

 I updated linux kernal to the version (chipyard:main firemarshal pointer), it can boot and login!
Thank you.

closed 2024-08-10 by franktaTian

jerryz123, 2024-08-10:
Glad to hear it!