ucb-bar/chipyard #1987: RoCC Punch Trough Signal

eric900115, 2024-08-09:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Hi all,

I am trying to develop a custom RoCC accelerator in rocket tile and would like to connect signals to a  master MMIO Accelerator and also to ChipTop. Are there any suggestions on how to punch through the signals from RoCC to both the MMIO Accelerator and ChipTop?

Thanks

### Motivating Example

Punch through the signals from RoCC to both the MMIO Accelerator and ChipTop

jerryz123, 2024-08-10:
I think the way to do this is 
1. Create a BundleBridgeSource in your rocc module with the signals that need to be punched to the MMIO accelerator
2. Create a BundleBridgeSink in your MMIO accelerator with the same signals
3. In your `CanHaveCustomMMIO` trait which integrates the MMIO into the system bus, query and filter the list of tiles and rocc-accelerators to find only your custom ones
4. Connect the BundleBridgeSource to the BundleBridgeSink using diplomacy's `:=`

I do something similar in rerocc, where the rerocc client (RoCC accelerators) punch signals to the ReRoCC bus (which just lives in the subsystem). This doesn't use BundleBriidgeSource, it uses a custom diplomacy node implementation, but the principle is the same.

See: 
 - https://github.com/ucb-bar/rerocc/blob/7281a84a6c1ac9165b886eb490a2e1cc97ee13c7/src/main/scala/Integration.scala#L24-L31
 - https://github.com/ucb-bar/rerocc/blob/7281a84a6c1ac9165b886eb490a2e1cc97ee13c7/src/main/scala/Integration.scala#L64-L79

eric900115, 2024-08-10:
Thanks, @jerryz123 . I will give it a try and would like to share my implementation here after I'm done.

franktaTian, 2024-08-26:
Or, for simple, just "How to punch rocc interface signals to ChipTop ,
so that one can integrate his co-processor just in verilog?

franktaTian, 2024-08-31:
Hi,
 I follow your guide step by step , but cannot find the "BundleBridgeSource". I think the problem is how to "query and filter the list of tiles and rocc-accelerators to find only your custom ones"  in CanHaveCustomMMIO.

> I think the way to do this is
>
> 1. Create a BundleBridgeSource in your rocc module with the signals that need to be punched to the MMIO accelerator
> 2. Create a BundleBridgeSink in your MMIO [...]

franktaTian, 2024-09-03:
Any help?

franktaTian, 2024-09-13:
> Thanks, @jerryz123 . I will give it a try and would like to share my implementation here after I'm done.

Do you have any progress?