ucb-bar/chipyard #2011: cospike_cosim not checking wdata for validity

pointerliu, 2024-08-21:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

main

### OS Setup

Linux pc 6.2.0-37-generic #38~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC x86_64 x86_64 x86_64 GNU/Linux

### Other Setup

I'm trying to add support cosim for CVA6. So I add a new config like this:
```scala
class CVA6CosimConfig extends Config(
  new chipyard.harness.WithCospike ++                   // attach spike-cosim
  new chipyard.config.WithTraceIO ++                    // enable the traceio
  new cva6.WithNCVA6Cores(1) ++
  new chipyard.config.AbstractConfig)
```
Then do `make CONFIG=CVA6CosimConfig` in `sims/verilator`. The object is generated successfully. However, when I try to cosim a elf file like this: `./simulator-chipyard.harness-CVA6CosimConfig $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple`.

The cosim return an error about mismatch on writing register. I check the CVA6 top file. It seems that the trace from CVA6 does not contain wdata signal. And `chipyard` also sets [traceHasWdata](https://github.com/ucb-bar/cva6-wrapper/blob/a2584610225ab296c4fb77a66f70eec64908efd0/src/main/scala/cva6/CVA6Tile.scala#L79) to be false.

### Current Behavior

cosim report a mismatch on registers. However, the wdata is not valid `has_wdata = false`.

And `has_wdata` is unused in function [cospike_cosim](https://github.com/ucb-bar/testchipip/blob/c94c1e3fa9f7437a3e95f63181cf0f54b8650b3a/src/main/resources/testchipip/csrc/cospike_impl.cc#L158)

### Expected Behavior

[cospike_cosim](https://github.com/ucb-bar/testchipip/blob/c94c1e3fa9f7437a3e95f63181cf0f54b8650b3a/src/main/resources/testchipip/csrc/cospike_impl.cc#L158) should not check wdata when `has_wata` is false.

### Other Information

If I just remove the check for wdata, the cosim for CVA6 also fails.
Now I don't know how to solve this problem, am I missing some features of CVA6? Can you give me some advice on CVA6 cosim?
```
Cosim: Configuring spike cosim
Cosim: isa string: rv64imafdczicsr_zifencei_zihpm_zicntr
Cosim: priv: MSU
Cosim: pmpregions: 8
Cosim: harts: 1
Cosim: htif args: /home/lzz/hAPR/chipyard/.conda-env/riscv-tools/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple
Cosim: Matching spike memory initial state for region 80000000-90000000
Cosim: Setting up htif
Cosim: Started
Cosim: Tohost addr   : 80001000
Cosim: Fromhost addr : 80001040
Cosim: BootROM base  : 10000
Cosim: BootROM size  : 10000
Cosim: Memory0 base  : 80000000
Cosim: Memory0 size  : 10000000
Cosim: Memory1 base  : 0
Cosim: Memory1 size  : 0
Cosim: Memory2 base  : 0
Cosim: Memory2 size  : 0
Cosim: 271 commit: 10000
core   0: 3 0x0000000000010000 (0x00000517) x10 0x0000000000010000
Cosim: 10f wdata mismatch reg 10 10000 != 0
Cosim: 281 commit: 10004
core   0: 3 0x0000000000010004 (0x04050513) x10 0x0000000000010040
Cosim: 119 wdata mismatch reg 10 10040 != 0
Cosim: 291 commit: 10008
core   0: 3 0x0000000000010008 (0x30551073) c773_mtvec 0x0000000000010040
Cosim: 301 commit: 1000c
core   0: 3 0x000000000001000c (0x301022f3) x5  0x800000000014112d
Cosim: CSR read 301
Cosim: CSR override: old=800000000014112d new=0
Cosim: 311 commit: 10010
core   0: 3 0x0000000000010010 (0x4122d293) x5  0x0000000000000000
Cosim: 321 commit: 10014
core   0: 3 0x0000000000010014 (0x0012f293) x5  0x0000000000000000
Cosim: 331 commit: 10018
core   0: 3 0x0000000000010018 (0x00028863)
Cosim: 341 commit: 1001c
core   0: 3 0x0000000000010028 (0x00800513) x10 0x0000000000000008
Cosim: 341 PC mismatch spike 10028 != DUT 1001c
```

jerryz123, 2024-08-21:
CVA6 cosim is not expected to work. There likely has to be some additional work done to CVA6 such that it exposes the trace with wdata in  the format cospike expects.

pointerliu, 2024-08-21:
Thanks ! I will check it.

As for the bug, do we need to fix it? If not, I will close the issue.

jerryz123, 2024-08-21:
If you have an interest in running cosim with CVA6, I'd be happy to review any PRs towards supporting that use case. Otherwise, I don't think anyone plans on fixing this at the moment.

P.S.: The CVA6 implementation has not been updated to the recent CVA6 RTL. That would need to be done before any additional work is done on CVA6,.

closed 2024-08-21 by pointerliu