ucb-bar/chipyard #2016: Build-setup issue

alsrbok, 2024-08-26:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`
x86_64 GNU/Linux

LSB Version:    :core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID: RedHatEnterprise
Description:    Red Hat Enterprise Linux release 8.10 (Ootpa)
Release:        8.10
Codename:       Ootpa

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

**Build-setup issue**

When we clone new version and try to run build-setup.sh with conda env, it failed.
1. build-setup.sh try to use .conda-lock-env, not .conda-env. This is mismatch with generated env.sh file/
2. There are changes on init-submodules-no-riscv-tools-nolog.sh file.
The current version does not seem to download some required executables from .conda-env/bin.
[init-submodules-no-riscv-tools-nolog.csv](https://github.com/user-attachments/files/16742889/init-submodules-no-riscv-tools-nolog.csv)
 I upload an older version that I have utilized successfully in the past.
 I resolved this problem by manually fixing lots of codes and files, but I hope we can correctly use it without any modification. 
 
Also, I upload the full version of log.
I think you can test it by cloning the project on new folder.

[log.docx](https://github.com/user-attachments/files/16742992/log.docx)

### Expected Behavior

.

jerryz123, 2024-08-26:
Can you upload the logs as a raw text file instead of .docx?

alsrbok, 2024-08-26:
$ git clone https://github.com/ucb-bar/chipyard.git
Cloning into 'chipyard'...
remote: Enumerating objects: 40993, done.
remote: Counting objects: 100% (1046/1046), done.
remote: Compressing objects: 100% (557/557), done.
remote: Total 40993 (delta 534), reused 825 (delta 422), pack-reused 39947 (from 1)
Receiving objects: 100% (40993/40993), 51.82 MiB | 36.85 MiB/s, done.
Resolving deltas: 100% (23062/23062), done.
$ conda create -n base_vpu python=3.10
(base_vpu) $ conda install conda-libmamba-solver  
(base_vpu) $ conda config --set solver libmamba
(base_vpu) $ ./build-setup.sh riscv-tools
 ========== BEGINNING STEP 1: Conda environment setup ==========                                                                                  
Channels:                                                                                                                                         
 - conda-forge                                                                                                                                    
 - defaults                                                                                                                                       
Platform: linux-64                                                                                                                                
Collecting package metadata (repodata.json): ...working... done                                                                                   
Solving environment: ...working... done                                                                                                                                                                                                                         
## Package Plan ##                                                                                                                                                                                                                                                    
  environment location: /home/ mgpark/chipyard-test/chipyard/.conda-lock-env                                                                                                                                                            
  added / updated specs:                                                                                                                          
    - conda-lock=2.5.7                                                                                                                            
  Downloading and Extracting Packages: ...working... done
Preparing transaction: ...working... done
Verifying transaction: ...working... done
Executing transaction: ...working... done
#
# To activate this environment, use
#
#     $ conda activate /home/ mgpark/chipyard-test/chipyard/.conda-lock-env
#
# To deactivate an active environment, use
#
#     $ conda deactivate

Using lockfile for conda: /home/ mgpark/chipyard-test/chipyard/conda-reqs/conda-lock-reqs/conda-requirements-riscv-tools-linux-64.conda-lock.yml
Storing main conda environment in /home/ mgpark/chipyard-test/chipyard/.conda-env
Usage: conda-lock install [OPTIONS] [LOCK_FILE]
Try 'conda-lock install --help' for help.

Error: Got unexpected extra arguments ({ \local cmd="${1-__missing__}"; case "$cmd" in activate | deactivate) __conda_activate "$@" ;; install | update | upgrade | remove | uninstall) __conda_exe "$@" || \return; __conda_reactivate ;; *) __conda_exe "$@" ;; esac } /home/ mgpark/chipyard-test/chipyard/conda-reqs/conda-lock-reqs/conda-requirements-riscv-tools-linux-64.conda-lock.yml)
build-setup.sh: Build script failed with exit code 2 at step 1: Conda environment setup

here it is.

abejgonzalez, 2024-08-26:
The error is unclear still. Can you run `build-setup.sh` with the `-v` flag for more output. (Also, you shouldn't have to create a new environment (`base_vpu`) to run `build-setup.sh`). My 1st guess is that there might be some version mismatch between `conda-lock` and `conda` but I don't know how that would happen. You could try running the `conda-lock` command yourself to debug as well:

```
conda-lock install --conda $(which conda) -p $CYDIR/.conda-env $CYDIR/conda-reqs/conda-lock-reqs/conda-requirements-$TOOLCHAIN_TYPE-linux-64.conda-lock.yml
```

https://github.com/ucb-bar/chipyard/blob/51b66dddb99146eacab3e5c7864e0fbc86859125/scripts/build-setup.sh#L190C5-L190C71 is where we create the `conda` environment with `conda-lock`.

alsrbok, 2024-08-26:
Thanks for your help. I think I resolve this issue.
I'll explain the workaround I used for other users

In my case few things interrupted me to succesfully run ./build-setup.sh riscv-tools

1) Change to **GLOBAL_ENV_NAME=".conda-env"** from GLOBAL_ENV_NAME=""
2) Utilize **CONDA_ENV_ARG="-p $GLOBAL_ENV_NAME"** instaed of CONDA_ENV_ARG="-n $GLOBAL_ENV_NAME"
(make sure to use -p option)
3) **Remove --conda $(which conda)** from conda-lock install
(In my case 'which conda' made unexpected argement error)
3) Utilize **conda activate .conda-env/** instead of conda activate $CONDA_ENV_NAME
(In my case it gave wrong command, leading to error)

Overall, check log whether they use proper command