ucb-bar/chipyard #2035: Error while executing "make buildfile CONFIG=GemminiRocketConfig tutorial=sky130-openroad"

chihyu-box, 2024-09-07:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: latest
Hash: 3705580bd70a8cfcbb7d27d3badb839938958706

### OS Setup

Linux yu-virtual-machine 6.8.0-40-generic #40~22.04.3-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 30 17:30:19 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### Other Setup

N/A

### Current Behavior

Executing `make buildfile CONFIG=GemminiRocketConfig tutorial=sky130-openroad` raised the following error.
Is this error caused by firtool not supporting this kind of expression?
```
generators/gemmini/src/main/scala/gemmini/DMA.scala:560:32: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/DMA.scala:560:32: note: see current operation: %1313 = "comb.mux"(%1002, %1311, %1312) <{twoState}> : (i1, !hw.array<4xi5>, !hw.array<4xi5>) -> !hw.array<4xi5>
generators/gemmini/src/main/scala/gemmini/DMA.scala:560:32: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/DMA.scala:560:32: note: see current operation: %1313 = "comb.mux"(%1002, %1311, %1312) <{twoState}> : (i1, !hw.array<4xi5>, !hw.array<4xi5>) -> !hw.array<4xi5>
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: note: see current operation: %1312 = "hw.array_create"(%1148, %1146, %1144, %1142) : (i5, i5, i5, i5) -> !hw.array<4xi5>
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: note: see current operation: %1311 = "comb.mux"(%928, %1307, %1310) <{twoState}> : (i1, !hw.array<4xi5>, !hw.array<4xi5>) -> !hw.array<4xi5>
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: note: see current operation: %1310 = "comb.mux"(%886, %1308, %1309) <{twoState}> : (i1, !hw.array<4xi5>, !hw.array<4xi5>) -> !hw.array<4xi5>
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: note: see current operation: %1310 = "comb.mux"(%886, %1308, %1309) <{twoState}> : (i1, !hw.array<4xi5>, !hw.array<4xi5>) -> !hw.array<4xi5>
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: note: see current operation: %1309 = "hw.array_create"(%469, %461, %452, %444) : (i5, i5, i5, i5) -> !hw.array<4xi5>
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: note: see current operation: %1310 = "comb.mux"(%886, %1308, %1309) <{twoState}> : (i1, !hw.array<4xi5>, !hw.array<4xi5>) -> !hw.array<4xi5>
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: note: see current operation: %1308 = "hw.array_create"(%621, %613, %605, %594) : (i5, i5, i5, i5) -> !hw.array<4xi5>
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: note: see current operation: %1311 = "comb.mux"(%928, %1307, %1310) <{twoState}> : (i1, !hw.array<4xi5>, !hw.array<4xi5>) -> !hw.array<4xi5>
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/DMA.scala:466:10: note: see current operation: %1307 = "hw.array_create"(%885, %874, %860, %845) : (i5, i5, i5, i5) -> !hw.array<4xi5>
src/main/scala/chisel3/util/Mux.scala:50:70: error: unsupported packed array expression
src/main/scala/chisel3/util/Mux.scala:50:70: note: see current operation: %1225 = "comb.mux"(%1002, %1223, %1224) <{twoState}> : (i1, !hw.array<4xi1>, !hw.array<4xi1>) -> !hw.array<4xi1>
src/main/scala/chisel3/util/Mux.scala:50:70: error: unsupported packed array expression
src/main/scala/chisel3/util/Mux.scala:50:70: note: see current operation: %1225 = "comb.mux"(%1002, %1223, %1224) <{twoState}> : (i1, !hw.array<4xi1>, !hw.array<4xi1>) -> !hw.array<4xi1>
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: note: see current operation: %1224 = "hw.array_create"(%1134, %1102, %1070, %1038) : (i1, i1, i1, i1) -> !hw.array<4xi1>
src/main/scala/chisel3/util/Mux.scala:50:70: error: unsupported packed array expression
src/main/scala/chisel3/util/Mux.scala:50:70: note: see current operation: %1225 = "comb.mux"(%1002, %1223, %1224) <{twoState}> : (i1, !hw.array<4xi1>, !hw.array<4xi1>) -> !hw.array<4xi1>
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: note: see current operation: %1223 = "hw.array_create"(%994, %978, %962, %946) : (i1, i1, i1, i1) -> !hw.array<4xi1>
src/main/scala/chisel3/util/Mux.scala:50:70: error: unsupported packed array expression
src/main/scala/chisel3/util/Mux.scala:50:70: note: see current operation: %1221 = "comb.mux"(%1002, %1219, %1220) <{twoState}> : (i1, !hw.array<4xi1>, !hw.array<4xi1>) -> !hw.array<4xi1>
src/main/scala/chisel3/util/Mux.scala:50:70: error: unsupported packed array expression
src/main/scala/chisel3/util/Mux.scala:50:70: note: see current operation: %1221 = "comb.mux"(%1002, %1219, %1220) <{twoState}> : (i1, !hw.array<4xi1>, !hw.array<4xi1>) -> !hw.array<4xi1>
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: note: see current operation: %1220 = "hw.array_create"(%1132, %1100, %1068, %1036) : (i1, i1, i1, i1) -> !hw.array<4xi1>
src/main/scala/chisel3/util/Mux.scala:50:70: error: unsupported packed array expression
src/main/scala/chisel3/util/Mux.scala:50:70: note: see current operation: %1221 = "comb.mux"(%1002, %1219, %1220) <{twoState}> : (i1, !hw.array<4xi1>, !hw.array<4xi1>) -> !hw.array<4xi1>
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: error: unsupported packed array expression
generators/gemmini/src/main/scala/gemmini/Util.scala:91:8: note: see current operation: %1219 = "hw.array_create"(%993, %977, %961, %945) : (i1, i1, i1, i1) -> !hw.array<4xi1>
src/main/scala/chisel3/util/Mux.scala:50:70: error: unsupported packed array expression
src/main/scala/chisel3/util/Mux.scala:50:70: note: see current operation: %1217 = "comb.mux"(%1002, %1215, %1216) <{twoState}> : (i1, !hw.array<4xi1>, !hw.array<4xi1>) -> !hw.array<4xi1>
src/main/scala/chisel3/util/Mux.scala:50:70: error: unsupported packed array expression
src/main/scala/chisel3/util/Mux.scala:50:70: note: see current operation: %1217 = "comb.mux"(%1002, %1215, %1216) <{twoState}> : (i1, !hw.array<4xi1>, !hw.array<4xi1>) -> !hw.array<4xi1>
[... 276 lines cut ...]
    0.5563 (  0.9%)    0.2790 (  0.7%)  'hw.module' Pipeline
    0.2365 (  0.4%)    0.1216 (  0.3%)    HWLegalizeModules
    0.2379 (  0.4%)    0.1564 (  0.4%)    PrettifyVerilog
    0.2864 (  0.5%)    0.2864 (  0.7%)  Rest
   59.4593 (100.0%)   38.9539 (100.0%)  Total
make: *** No rule to make target '/home/yu/chipyard_2/vlsi/generated-src/chipyard.harness.TestHarness.GemminiRocketConfig/model_module_hierarchy.json', needed by '/home/yu/chipyard_2/vlsi/generated-src/chipyard.harness.TestHarness.GemminiRocketConfig/chipyard.harness.TestHarness.GemminiRocketConfig.top.f'.  Stop.
```

### Expected Behavior

N/A

### Other Information

Reproduce the error:
```
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
git checkout 3705580bd70a8cfcbb7d27d3badb839938958706
./build-setup.sh riscv-tools
source env.sh
cd vlsi
# Add the tool and technology paths to the example-sky130.yml and example-openroad.yml files.
make buildfile CONFIG=GemminiRocketConfig tutorial=sky130-openroad
```

jerryz123, 2024-09-07:
The tutorial flow is trying to support Yosys, which does not support SV packed arrays. However, the CIRCT `disallowPackedArrays` option seems to be broken, as it causes these errors.

chihyu-box, 2024-09-08:
Hi Jerry, I noticed that if I remove `disallowPackedArray` option in `common.mk`, the error no longer occur. However, it seems like the `sram-cache.json` for `sky130` is missing an sutiable SRAM for `mem_0_ext`, which is causing the following error. 

Can sram22 generate a suitable SRAM for `mem_0_ext`?

Content of  `chipyard.harness.TestHarness.GemminiRocketConfig.top.mems.conf` 
```
name cc_dir_ext depth 1024 width 136 ports mrw mask_gran 17 
name cc_banks_0_ext depth 8192 width 64 ports rw 
name rockettile_dcache_data_arrays_0_ext depth 256 width 512 ports mrw mask_gran 8 
name rockettile_dcache_tag_array_ext depth 64 width 176 ports mrw mask_gran 22 
name mem_ext depth 4096 width 128 ports mrw mask_gran 8 
name mem_0_ext depth 512 width 512 ports mwrite,read mask_gran 8 
name rockettile_icache_tag_array_ext depth 64 width 168 ports mrw mask_gran 21 
name rockettile_icache_data_arrays_0_ext depth 256 width 256 ports mrw mask_gran 32 
name mem_1_ext depth 8192 width 64 ports mrw mask_gran 8 
```
Error log
```
cd /home/yu/chipyard_2 && java -cp /home/yu/chipyard_2/.classpath_cache/tapeout.jar tapeout.macros.MacroCompiler -n /home/yu/chipyard_2/vlsi/generated-src/chipyard.harness.TestHarness.GemminiRocketConfig/chipyard.harness.TestHarness.GemminiRocketConfig.top.mems.conf -v /home/yu/chipyard_2/vlsi/generated-src/chipyard.harness.TestHarness.GemminiRocketConfig/gen-collateral/chipyard.harness.TestHarness.GemminiRocketConfig.top.mems.v -f /home/yu/chipyard_2/vlsi/generated-src/chipyard.harness.TestHarness.GemminiRocketConfig/chipyard.harness.TestHarness.GemminiRocketConfig.top.mems.fir -l  /home/yu/chipyard_2/.conda-env/lib/python3.10/site-packages/hammer/technology/sky130/sram-cache.json -hir /home/yu/chipyard_2/vlsi/generated-src/chipyard.harness.TestHarness.GemminiRocketConfig/chipyard.harness.TestHarness.GemminiRocketConfig.mems.hammer.json --mode strict
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/yu/chipyard_2/.java_tmp
Unmasked target memory: unaligned mem maskGran 17 with lib (sram22_4096x8m8w8) width 8 not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 17 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 17 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 17 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 17 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 17 is not a multiple of lib maskGran 2: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 17 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 17 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 17 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Unmasked target memory: unaligned mem maskGran 22 with lib (sram22_4096x8m8w8) width 8 not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 22 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 22 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 22 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 22 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 22 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 22 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
Mem maskGran 22 is not a multiple of lib maskGran 8: currently not supported
Error occurred during bitPairs calculations (bitPairs is empty).
INFO: unable to compile mem_0_ext using sram22_4096x8m8w8 port count must match
INFO: unable to compile mem_0_ext using sram22_512x32m4w32 port count must match
INFO: unable to compile mem_0_ext using sram22_64x32m4w32 port count must match
INFO: unable to compile mem_0_ext using sram22_512x64m4w8 port count must match
INFO: unable to compile mem_0_ext using sram22_512x32m4w8 port count must match
INFO: unable to compile mem_0_ext using sram22_64x24m4w24 port count must match
INFO: unable to compile mem_0_ext using sram22_4096x32m8w8 port count must match
INFO: unable to compile mem_0_ext using sram22_256x32m4w8 port count must match
INFO: unable to compile mem_0_ext using sram22_64x4m4w2 port count must match
INFO: unable to compile mem_0_ext using sram22_2048x32m8w8 port count must match
INFO: unable to compile mem_0_ext using sram22_1024x32m8w32 port count must match
INFO: unable to compile mem_0_ext using sram22_1024x32m8w8 port count must match
INFO: unable to compile mem_0_ext using sram22_64x32m4w8 port count must match
Options:
[... cut ...]
	at scala.App.main$(App.scala:96)
	at tapeout.macros.MacroCompiler$.main(MacroCompiler.scala:787)
	at tapeout.macros.MacroCompiler.main(MacroCompiler.scala)
make: *** No rule to make target '/home/yu/chipyard_2/vlsi/generated-src/chipyard.harness.TestHarness.GemminiRocketConfig/gen-collateral/chipyard.harness.TestHarness.GemminiRocketConfig.top.mems.v', needed by '/home/yu/chipyard_2/vlsi/generated-src/chipyard.harness.TestHarness.GemminiRocketConfig/syn.f'.  Stop.
```
Can sram22 generate a suitable SRAM for `mem_0_ext`?