ucb-bar/chipyard #2046: How to export my custom IO signals at RocketTile  module to DigitalTop's IO?

csgxiong, 2024-09-14:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

I have some custom IO signals (Both input and output ) at RocketTile module.And I want to export them to  DigitalTop's IO.

### Motivating Example

I have some custom IO signals (Both input and output ) at RocketTile module.And I want to export them to  DigitalTop's IO.
When I try to  add codes below, it will cause some errors.
Code:
class ChipyardSubsystemModuleImp[+L <: ChipyardSubsystem](_outer: L) extends BaseSubsystemModuleImp(_outer)
  with HasTilesModuleImp
{
  // Generate C header with relevant information for Dromajo
  // This is included in the `dromajo_params.h` header file
  DromajoHelper.addArtefacts(InSubsystem)
  val io = IO(new Bundle{
     val clk_en = Output(Bool())
  })
  io.clk_en := outer.tiles(0).module.roccPort.clk_en
}
Error log :
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at ... ()
	at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36)
	at ... ()
	at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
Caused by: chisel3.internal.ChiselException: Connection between sink (DigitalTop.io.clk_en: IO[Bool]) and source (RocketTile.roccPort.clk_en: IO[Bool]) failed @: Sink or source unavailable to current module.
	at chisel3.internal.throwException$.apply(Error.scala:169)
	at chisel3.Data.connect(Data.scala:635)
	at chisel3.Data.$anonfun$$colon$eq$1(Data.scala:796)
	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.scala:18)
	at chisel3.internal.prefix$.apply(prefix.scala:31)
	at chisel3.Data.$colon$eq(Data.scala:796)

So Does anyone have any ideas on this?
Thanks.
B.R.

jerryz123, 2024-09-14:
You should create a new trait for DigitalTop named: `CanHaveRocketTileCustomIO`. Within this trait, you can filter the `totalTiles` list to get the RocketTile modules. 

Within each RocketTile, you should create a BundleBridgeSource node as the "source" of your signal.
Within `CanHaveRocketTileCustomIO`, you can create a corresponding `BundleBridgeSink` for each RocketTile, and connect it to the matching RocketTile.

Then, in `CanHaveRocketTileCustomIO`, you'll have access to the bundle from the BundleBridgeSink. You can punch this signal out as a IO of DigitalTop

csgxiong, 2024-09-18:
@jerryz123 
Tank you, Jerry. What you have mentioned above is just like the way reset_vector / hart_id IO to be exported at HasTiles.scala.
I had tried this way.But errors happened as below:
ERROR: BundleBridge.scala:137 "can only be used with Outptut-Directed Bundles"

My custom IO signals Not only have Input  Directed IO, but also have  Output Directed IO.So it seems to find another way.
I am not very familiar with the Chipyard project.

Thanks.
B.R.