ucb-bar/chipyard #2047: the data writing to SIMDRAM is always 0

xlgforever, 2024-09-14:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.11.0

### OS Setup

Linux xlg16p 6.8.0-40-generic #40~22.04.3-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 30 17:30:19 UTC 2 x86_64 GNU/Linux

Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

I have test the `make run-binary-debug BINARY=symmetric.riscv` in the `sims/verilator` folder.  and the simulation have finished successfully:
```
Running with RISCV=/home/xlg/wrk/chipyard/.conda-env/riscv-tools
if [ "symmetric.riscv" != "none" ] && [ ! -f "symmetric.riscv" ]; then printf "\n\nBinary symmetric.riscv not found\n\n"; exit 1; fi
if [ "symmetric.riscv" != "none" ]; then riscv64-unknown-elf-objdump -D -S symmetric.riscv > /home/xlg/wrk/chipyard/sims/verilator/output/chipyard.harness.TestHarness.MyRocketConfig/symmetric.dump ; fi
(set -o pipefail &&  /home/xlg/wrk/chipyard/sims/verilator/simulator-chipyard.harness-MyRocketConfig-debug +permissive +dramsim +dramsim_ini_dir=/home/xlg/wrk/chipyard/generators/testchipip/src/main/resources/dramsim2_ini +max-cycles=10000000     +verbose +vcdfile=/home/xlg/wrk/chipyard/sims/verilator/output/chipyard.harness.TestHarness.MyRocketConfig/symmetric.vcd +permissive-off symmetric.riscv </dev/null 2> >(spike-dasm > /home/xlg/wrk/chipyard/sims/verilator/output/chipyard.harness.TestHarness.MyRocketConfig/symmetric.out) | tee /home/xlg/wrk/chipyard/sims/verilator/output/chipyard.harness.TestHarness.MyRocketConfig/symmetric.log)
[UART] UART0 is here (stdin/stdout).
Wrote 94 bytes in 247 cycles
Read 94 bytes in 164 cycles
- /home/xlg/wrk/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.MyRocketConfig/gen-collateral/TestDriver.v:158: Verilog $finish
```

the `symmetric.riscv` case( which is generated by the `tests/symmetric.c`) shoud wring some string to off-chip mem, and read back. 

### Expected Behavior

the simulation seems to be run successfully according to the output of uart in simulation. However, when I use gtkwave to check the signals of AXI4 of `SIMDRAM`, the axi_r_data is always 0( but the axi_r_valid & axi_r_ready are assserted for some times).
Besides, the axi_w_valid & axi_w_ready are only asserted for one time. Does this mean that there is no data written by the cpu in simulation? 
The signal waves are not matched to the simulating case.

jerryz123, 2024-09-14:
When running a RTL simulation with `run-binary`, the binary is written into the front port of the SoC, which writes directly into the L2. The system reads in the "dirty" data from DRAM, which is initialized to zeros, then caches it in the L2. So you never see the program data going over the AXI port.

If you run with the `LOADMEM=1` flag, the binary will be written into DRAM initially, so the first loads to DRAM performed by the SoC should read the program data.

xlgforever, 2024-09-15:
> When running a RTL simulation with `run-binary`, the binary is written into the front port of the SoC, which writes directly into the L2. The system reads in the "dirty" data from DRAM, which is [...]

But in the symmetric program, what it does is write a string to the DDR, then read the string back from the DDR, and verify it. Isn't the operation of wrting ddr  also invisible on the SIMDDR  axi bus? I didn't observe writing operation in the simulated waveform.

jerryz123, 2024-09-15:
It is likely because the DDR is cached into the SoC, so you won't see the write traffic going out to SimDRAM unless there is a cache eviction. This is the expected behavior.

xlgforever, 2024-09-16:
> It is likely because the DDR is cached into the SoC, so you won't see the write traffic going out to SimDRAM unless there is a cache eviction. This is the expected behavior.

I see, thank you！

xlgforever, 2024-09-16:
> It is likely because the DDR is cached into the SoC, so you won't see the write traffic going out to SimDRAM unless there is a cache eviction. This is the expected behavior.

I see, thank you

closed 2024-09-16 by xlgforever