ucb-bar/chipyard #2065: ./build-setup.sh riscv-tools error

luoqi-1, 2024-09-30:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

during ./build-setup.sh riscv-tools
step1:initializing Conda environment
I got error like this
how can I solve it?

### Expected Behavior

 d

### Other Information

To activate this environment, use
INFO:root:#
INFO:root:#     $ conda activate /home/ICer/work_space/chipyard/.conda-env
INFO:root:#
INFO:root:# To deactivate an active environment, use
INFO:root:#
INFO:root:#     $ conda deactivate
INFO:root:
ERROR:root:ERROR: bcrypt-4.2.0-cp39-abi3-manylinux_2_28_x86_64.whl is not a supported wheel on this platform.
ERROR:root:
ERROR:root:ERROR conda.cli.main_run:execute(125): `conda run pip install --no-deps -r /tmp/tmp6ydxfff5` failed. (See above for error)
Traceback (most recent call last):
  File "/home/ICer/work_space/chipyard/.conda-lock-env/bin/conda-lock", line 10, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/ICer/work_space/chipyard/.conda-lock-env/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ICer/work_space/chipyard/.conda-lock-env/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/ICer/work_space/chipyard/.conda-lock-env/lib/python3.12/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ICer/work_space/chipyard/.conda-lock-env/lib/python3.12/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ICer/work_space/chipyard/.conda-lock-env/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ICer/work_space/chipyard/.conda-lock-env/lib/python3.12/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ICer/work_space/chipyard/.conda-lock-env/lib/python3.12/site-package

jerryz123, 2024-10-03:
Centos 7.9 is EOL now. I suspect the error has to do with that

suhwanhong999, 2024-11-08:
Have you tried with conda-lock version 1.4.0? Seems that a newer version of conda-lock causes problems for me when running ./build-setup.sh

haipnh, 2024-11-11:
@suhwanhong999 I followed [Initial-Repo-Setup](https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Initial-Repo-Setup.html) and installed conda-lock version 1.4.0, but the error occurred same as @ZechenM 's.

haipnh, 2024-11-11:
@ZechenM May you try deactivating base conda, then add conda/bin path to your $PATH? It worked on my machine.

ZechenM, 2024-11-12:
@suhwanhong999 @haipnh thank you both for your inputs! I ended up using an older version of firesim (1.17.1), which sets up my project successfully