ucb-bar/chipyard #2091: Error: Failed to make rmapats.so

RockyQLuo, 2024-10-21:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6

### OS Setup

Ex: Output of `uname -a` + `lsb_release -a` + `printenv` + `conda list`

Linux qluo-u 6.8.0-47-generic #47~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Oct  2 16:16:55 UTC 2 x86_64 GNU/Linux

LSB Version:    core-11.1.0ubuntu4-noarch:printing-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

after running:
cd chipyard/sims/vcs
make SUB_PROJECT=constellation BINARY=none CONFIG=TestConfig00 run-binary-debug
then i got an error like :
Error: Failed to make rmapats.so @ /home/qluo/workspace/chipyard/sims/vcs/simv-constellation.test-TestConfig00-debug.daidir/rmapats.so

and i check that  rmapats.so
[image]

### Expected Behavior

 the generated verilog will be in generated-src.

jerryz123, 2024-10-28:
This feels like an issue with the system VCS configuration. What happens if you try to build the default config (Just run `make`) in `sims/vcs`?

Cain252199, 2024-11-10:
My system is also Ubuntu 20.04 LTS, and the Chipyard version is 1.13.0. I also tried to run **make**, but it failed. Have you resolved this issue?
[image]