ucb-bar/chipyard #2098: How to configure asynchronous reset

AD738560581, 2024-10-25:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

Hello, I have encountered some questions and would like to seek advice. At present, the RTL I generate under chipyard is synchronously reset. I hope to generate an asynchronous reset system. Are there any configuration options that can be modified？

### Motivating Example

Hello, I have encountered some questions and would like to seek advice. At present, the RTL I generate under chipyard is synchronously reset. I hope to generate an asynchronous reset system. Are there any configuration options that can be modified？

jerryz123, 2024-10-28:
The reset pin into the ChipTop is asynchronous w.r.t. any other clock inputs. A reset synchronizer sync's that reset to the clock for the digital logic.

AD738560581, 2024-10-29:
Thank you for your reply, but if I wish to remove the synchronization tool for chiptop. Because I hope to ultimately obtain a digital top system with asynchronous reset. 
I added the option "with RequireAsyncReset" in the digital configuration, but there were a series of errors, such as "reset network reset simultaneously connected to async and sync". Do you have any suggestions to solve this problem？

jerryz123, 2024-10-29:
This isn't well supported anymore.  To support the complexity of the clocking and reset network for a multi-clock-domain SoC, it was necessary to move to global synchronous reset. Reverting that would be a very invasive change.

Could you describe why you need asynchronous reset for all the digital logic?

AD738560581, 2024-10-30:
Because our project requires this project as a sub IP, but we only use the Digital level
However, other IPs are asynchronous resets, and we hope to achieve global asynchronous resets, including Digital top

AD738560581, 2024-10-30:
Also, may I ask if I can generate a system with Digital as the top-level separately

commit 3bdabbe6 (2025-05-15): Add support for async reset

Users can now use the `WithAsyncClockGroups` config fragment to coerce
certain clock groups to use an asynchronous reset. Note that users will
likely also want to use the new `WithAsyncResetRocketSubsystem` config
fragment, which ensures the Rocket Chip Debug Module also uses an async
reset. Omitting this will lead to signals being driven by both sync and
async resets, which will cause `firtool` compilation errors.

The new `AsyncResetRocketConfig` config demonstrates this. Note that it
doesn't use async reset for the Rocket Tile itself (just the uncore), as
the Ro [... cut]
closed 2025-05-15 by jerryz123