ucb-bar/chipyard #2102: i want to build a customed SOC bitstream in chipyard /FPGA

cinder023, 2024-10-28:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Feature Description

I want to generate my own custom SOC to burn a bitstream file to a supported development board (i.e. not limited to the off-the-shelf configuration provided by the config.scala file in the FPGA/src/main/scala/board folder) What configuration should I add?

### Motivating Example

For example, I want to generate a CVA6+AXI based SOC in Nexysvideo.However,there is no related configuration about CVA6 in the 
config.scala under FPGA/src/main/scala/board folder

cinder023, 2024-10-30:
> You'll have to construct the configuration yourself, matching what was done for the example configurations for that board

by the way,i want to combine CVA6 +ARA in chipyard ，I don't know how to start at the moment, should I use chisel to exemplify cva6 blackbox.scala and ara blackbox.scala?

jerryz123, 2024-10-30:
Ah, I did not realize you wanted to do CVA6+Ara. That is not supported in
CY at the moment, the CVA6 and Ara blocks are independent IPs that are not
combinable in the chisel wrappers.

The approach would be to create a new Chisel tile wrapper - "CVA6AraTile"
that wraps a combined CVA6+Ara system.

-Jerry