ucb-bar/chipyard #2119: basic design fail to meet timing and a really slow booting linux on VCU118

Charles-Tang, 2024-11-08:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.13.0
Hash:69eba860a352343e4ac6b6df0f3638a79a86ec78

### OS Setup

Linux charles-PC 6.6.47-amd64-desktop-hwe #23.01.00.37 SMP PREEMPT_DYNAMIC Tue Aug 20 17:33:52 CST 2024 x86_64 GNU/Linux + Distributor ID: Deepin Description: Deepin 23 Release: 23 Codename: beige + SHELL=/bin/bash WINDOWID=0 COLORTERM=truecolor XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0 GNOME_DESKTOP_SESSION_ID=this-is-deprecated GTK_IM_MODULE=fcitx CONDA_EXE=/home/charles/miniconda3/bin/conda _CE_M= GNOME_KEYRING_CONTROL=/run/user/1000/keyring LANGUAGE= D_DISABLE_RT_SCREEN_SCALE=1 XILINX_VIVADO=/opt/Xilinx/Vivado/2022.2 SSH_AUTH_SOCK=/run/user/1000/keyring/ssh XDG_DATA_HOME=/home/charles/.local/share XDG_CONFIG_HOME=/home/charles/.config MEMORY_PRESSURE_WRITE=c29tZSAyMDAwMDAgMjAwMDAwMAA= XMODIFIERS=@im=fcitx DESKTOP_SESSION=deepin XDG_SEAT=seat0 PWD=/home/charles/Documents/git/chipyard XDG_SESSION_DESKTOP=dde-x11 LOGNAME=charles XDG_SESSION_TYPE=x11 CONDA_PREFIX=/home/charles/miniconda3 GNOME_KEYRING_CUSTOM_LOCAL_PATH=/home/charles/.local/share/deepin-keyrings-wb GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1 SYSTEMD_EXEC_PID=3568 XAUTHORITY=/home/charles/.Xauthority XDG_GREETER_DATA_DIR=/var/lib/lightdm/data/charles HOME=/home/charles IM_CONFIG_PHASE=1 LANG=zh_CN.UTF-8 LS_COLORS=bd=38;5;68;1:ca=38;5;17:cd=38;5;132;1:di=38;5;105:do=38;5;127:ex=38;5;80:pi=38;5;126:fi=38;5;167:ln=38;5;63:mh=38;5;99;1:or=48;5;197;38;5;228;1:ow=38;5;220;1:sg=48;5;234;38;5;100;1:su=38;5;9;1:so=38;5;197:st=38;5;86;48;5;234:tw=48;5;235;38;5;139;3:*LS_COLORS=48;5;89;38;5;197;1;3;4;7:*.BAT=38;5;108:*.exe=38;5;196:*.PL=38;5;160:*.asm=38;5;240;1:*.awk=38;5;148;1:*.bash=38;5;173:*.bat=38;5;108:*.c=38;5;110:*.cc=38;5;24;1:*.ii=38;5;24;1:*.cfg=1:*.cl=38;5;204;1:*.coffee=38;5;100;1:*.conf=38;5;221:*.C=38;5;24;1:*.cp=38;5;24;1:*.cpp=38;5;24;1:*.cxx=38;5;24;1:*.c++=38;5;24;1:*.ii=38;5;24;1:*.cs=38;5;74;1:*.css=38;5;91:*.csv=38;5;78:*.ctp=38;5;95:*.diff=48;5;197;38;5;232:*.enc=38;5;21:*.eps=38;5;33;1:*.etx=38;5;172:*.ex=38;5;148;1:*.example=38;5;225;1:*.git=38;5;197:*.gitignore=38;5;240:*.go=38;5;111:*.h=38;5;81:*.H=38;5;81:*.hpp=38;5;81:*.hxx=38;5;81:*.h++=38;5;81:*.tcc=38;5;81:*.f=38;5;81:*.for=38;5;81:*.ftn=38;5;81:*.s=38;5;81:*.S=38;5;81:*.sx=38;5;81:*.hi=38;5;240:*.hs=38;5;199;1:*.htm=38;5;125;1:*.html=38;5;9;1:*.info=38;5;101:*.ini=38;5;123:*.java=38;5;142;1:*.jhtm=38;5;125;1:*.js=38;5;42:*.jsm=38;5;42:*.jsm=38;5;42:*.json=38;5;213:*.jsp=38;5;45:*.lisp=38;5;204;1:*.log=38;5;170:*.lua=38;5;34;1:*.m=38;5;130;3:*.mht=38;5;129:*.mm=38;5;130;3:*.M=38;5;130;3:*.map=38;5;58;3:*.markdown=38;5;184:*.md=38;5;184:*.mf=38;5;220;3:*.mfasl=38;5;73:*.mi=38;5;124:*.mkd=38;5;184:*.mtx=38;5;36;3:*.nfo=38;5;220:*.o=38;5;240:*.pacnew=38;5;33:*.patch=48;5;197;38;5;232;1:*.pc=38;5;100:*.pfa=38;5;43:*.pgsql=38;5;222:*.php=38;5;99:*.pl=38;5;214:*.po=38;5;176:*.pot=38;5;206;1:*.plt=38;5;204;1:*.pm=38;5;197;1:*.pod=38;5;172;1:*.py=38;5;41:*.pyc=38;5;245:*.rb=38;5;190:*.rdf=38;5;144:*.rst=38;5;67:*.ru=38;5;142:*.scm=38;5;204;1:*.sed=38;5;130;1:*.sfv=38;5;197:*.sh=38;5;113:*.signature=38;5;206:*.sql=38;5;222:*.srt=38;5;116:*.sty=38;5;58:*.sug=38;5;44:*.t=38;5;28;1:*.tcl=38;5;64;1:*.tdy=38;5;214:*.tex=38;5;172:*.textile=38;5;106:*.tfm=38;5;64:*.tfnt=38;5;140:*.theme=38;5;109:*.txt=38;5;33;1:*.list=38;5;44:*.save=38;5;240:*.urlview=38;5;85:*.vim=38;5;33;1:*.vimrc=38;5;13:*.viminfo=38;5;240;1:*.xml=38;5;199:*.yml=38;5;208:*.zsh=38;5;173:*.bashrc=38;5;5:*README=38;5;220;1:*Makefile=38;5;196:*MANIFEST=38;5;243:*pm_to_blib=38;5;240:*.1=38;5;240;1:*.2=38;5;220;1:*.3=38;5;196;1:*.7=38;5;196;1:*.1p=38;5;160:*.3p=38;5;160:*.am=38;5;242:*.in=38;5;242:*.old=38;5;242:*.out=38;5;44;1:*.bmp=38;5;33:*.cdr=38;5;33:*.gif=38;5;33:*.ico=38;5;33:*.jpeg=38;5;33:*.jpg=38;5;33:*.JPG=38;5;33:*.png=38;5;33:*.svg=38;5;33;1:*.xpm=38;5;33:*.32x=38;5;137:*.A64=38;5;82:*.a00=38;5;99:*.a52=38;5;112:*.a64=38;5;82:*.a78=38;5;112:*.adf=38;5;35:*.atr=38;5;213:*.cdi=38;5;124:*.fm2=38;5;35:*.gb=38;5;203:*.gba=38;5;205:*.gbc=38;5;204:*.gel=38;5;83:*.gg=38;5;138:*.ggl=38;5;83:*.j64=38;5;102:*.nds=38;5;199:*.nes=38;5;160:*.rom=38;5;59;1:*.sav=38;5;220:*.sms=38;5;33:*.st=38;5;208;1:*.iso=38;5;9;1:*.nrg=38;5;124:*.qcow=38;5;141:*.VOB=38;5;99:*.IFO=38;5;99:*.BUP=38;5;99:*.MOV=38;5;99:*.3gp=38;5;99:*.3g2=38;5;99:*.asf=38;5;99:*.avi=38;5;99:*.divx=38;5;99:*.f4v=38;5;99:*.flv=38;5;99:*.m2v=38;5;99:*.mkv=38;5;99:*.mov=38;5;99:*.mp4=38;5;99:*.mpg=38;5;99:*.mpeg=38;5;99:*.ogm=38;5;99:*.ogv=38;5;99:*.rmvb=38;5;99:*.sample=38;5;99;1:*.ts=38;5;99:*.vob=38;5;99:*.webm=38;5;99:*.wmv=38;5;99:*.S3M=38;5;69;1:*.aac=38;5;69:*.cue=38;5;69:*.dat=38;5;69:*.dts=38;5;69;1:*.fcm=38;5;69:*.flac=38;5;69;1:*.m3u=38;5;69:*.m3u8=38;5;69:*.m4=38;5;69;3:*.m4a=38;5;69;1:*.mid=38;5;69:*.midi=38;5;69:*.mod=38;5;69:*.mp3=38;5;69:*.oga=38;5;69:*.ogg=38;5;69:*.s3m=38;5;69;1:*.sid=38;5;69;1:*.spl=38;5;69:*.wv=38;5;69:*.wvc=38;5;69:*.ape=38;5;69:*.afm=38;5;58:*.pfb=38;5;58:*.pfm=38;5;58:*.ttf=38;5;66:*.ttc=38;5;66:*.pcf=38;5;65:*.psf=38;5;64:*.bak=38;5;222;1:*.bin=38;5;196:*.pid=38;5;33;1:*.state=38;5;21:*.swo=38;5;21:*.swp=38;5;21:*.tmp=38;5;21:*.un~=38;5;21:*.zcompdump=38;5;21:*.zwc=38;5;21:*.db=38;5;147:*.dump=38;5;147:*.sqlite=38;5;147:*.typelib=38;5;147:*.7z=38;5;140:*.a=38;5;220:*.apk=38;5;148;3:*.arj=38;5;220:*.bz2=38;5;220:*.deb=38;5;215;1:*.ipk=38;5;220:*.jad=38;5;220:*.jar=38;5;220:*.nth=38;5;220:*.sis=38;5;220:*.part=38;5;220;1:*.r00=38;5;220:*.r01=38;5;220:*.r02=38;5;220:*.r03=38;5;220:*.r04=38;5;220:*.r05=38;5;220:*.r06=38;5;220:*.r07=38;5;220:*.r08=38;5;220:*.r09=38;5;220:*.r10=38;5;220:*.r100=38;5;220:*.r101=38;5;220:*.r102=38;5;220:*.r103=38;5;220:*.r104=38;5;220:*.r105=38;5;220:*.r106=38;5;220:*.r107=38;5;220:*.r108=38;5;220:*.r109=38;5;220:*.r11=38;5;220:*.r110=38;5;220:*.r111=38;5;220:*.r112=38;5;220:*.r113=38;5;220:*.r114=38;5;220:*.r115=38;5;220:*.r116=38;5;220:*.r12=38;5;220:*.r13=38;5;220:*.r14=38;5;220:*.r15=38;5;220:*.r16=38;5;220:*.r17=38;5;220:*.r18=38;5;220:*.r19=38;5;220:*.r20=38;5;220:*.r21=38;5;220:*.r22=38;5;220:*.r25=38;5;220:*.r26=38;5;220:*.r27=38;5;220:*.r28=38;5;220:*.r29=38;5;220:*.r30=38;5;220:*.r31=38;5;220:*.r32=38;5;220:*.r33=38;5;220:*.r34=38;5;220:*.r35=38;5;220:*.r36=38;5;220:*.r37=38;5;220:*.r38=38;5;220:*.r39=38;5;220:*.r40=38;5;220:*.r41=38;5;220:*.r42=38;5;220:*.r43=38;5;220:*.r44=38;5;220:*.r45=38;5;220:*.r46=38;5;220:*.r47=38;5;220:*.r48=38;5;220:*.r49=38;5;220:*.r50=38;5;220:*.r51=38;5;220:*.r52=38;5;220:*.r53=38;5;220:*.r54=38;5;220:*.r99=38;5;220:*.r56=38;5;220:*.r69=38;5;220:*.r58=38;5;220:*.r59=38;5;220:*.r60=38;5;220:*.r61=38;5;220:*.r62=38;5;220:*.r63=38;5;220:*.r64=38;5;220:*.r65=38;5;220:*.r66=38;5;220:*.r67=38;5;220:*.r68=38;5;220:*.r69=38;5;220:*.r69=38;5;220:*.r70=38;5;220:*.r71=38;5;220:*.r72=38;5;220:*.r73=38;5;220:*.r74=38;5;220:*.r75=38;5;220:*.r76=38;5;220:*.r77=38;5;220:*.r78=38;5;220:*.r79=38;5;220:*.r80=38;5;220:*.r81=38;5;220:*.r82=38;5;220:*.r83=38;5;220:*.r84=38;5;220:*.r85=38;5;220:*.r86=38;5;220:*.r87=38;5;220:*.r88=38;5;220:*.r89=38;5;220:*.r90=38;5;220:*.r91=38;5;220:*.r92=38;5;220:*.r99=38;5;220:*.r94=38;5;220:*.r95=38;5;220:*.r96=38;5;220:*.r97=38;5;220:*.r98=38;5;220:*.r99=38;5;220:*.rar=38;5;217;1:*.tar=38;5;130;1:*.tar.gz=38;5;130;1:*.tgz=38;5;130;1:*.gz=38;5;130;1:*.xz=38;5;130;1:*.zip=38;5;130;1:*.doc=38;5;33;1:*.wps=38;5;33;1:*.docx=38;5;33;1:*.xls=38;5;33;1:*.xlsx=38;5;33;1:*.ppt=38;5;33;1:*.pptx=38;5;33;1:*.pdf=38;5;33;1:*.djvu=38;5;33;1:*.cbr=38;5;33;1:*.cbz=38;5;33;1:*.chm=38;5;33;1:*.odt=38;5;112:*.ods=38;5;112:*.odp=38;5;112:*.odb=38;5;112:*.allow=38;5;112:*.deny=38;5;203:*.SKIP=38;5;244:*.def=38;5;136:*.directory=38;5;69:*.err=38;5;160;1:*.error=38;5;160;1:*.pi=38;5;126:*.properties=38;5;197;1:*.torrent=38;5;105:*.gp3=38;5;114:*.gp4=38;5;115:*.tg=38;5;99:*.pcap=38;5;29:*.cap=38;5;29:*.dmp=38;5;29:*.service=38;5;81:*@.service=38;5;45:*.socket=38;5;75:*.device=38;5;24:*.mount=38;5;115:*.automount=38;5;114:*.swap=38;5;113:*.target=38;5;73:*.path=38;5;116:*.timer=38;5;111:*.snapshot=38;5;139:*.desktop=38;5;113:*.crdownload=38;5;38:*.crx=38;5;13: XDG_CURRENT_DESKTOP=DDE MEMORY_PRESSURE_WATCH=/sys/fs/cgroup/user.slice/user-1000.slice/user@1000.service/app.slice/app-DDE-deepin\x2dterminal@b66e067 [... cut ...]
### Current Behavior

i can success with default rocket configuration, it teminated with a failure of slack=-0.641, so i relax the frequency to 75MHz, and when i boot on VCU118 with the successfully generated bitstream, the output of remote terminal is :
INIT
CMD0
CMD8
ACMD41
CMD58
CMD16
CMD18
LOADING 0x01e00000 B PAYLOAD
LOADING -
and there are no more output

when i do the same in 1.8.0, i can boot successfuly within minutes, but now i cannot get pass this LOADING

### Expected Behavior

boot linux successfully

Charles-Tang, 2024-11-11:
i switch to 11.0 the stable verison and it seems work just fine, still no clue why these happens in 1.13, the configuration of chipyard and rocket-chip should be the same

wadidf, 2024-11-11:
Hi,
As a workaround for the timing problem. 
Revert back to 100Mhz it is fine but force the exploration of place & route:
- Edit the line 11 in "**./fpga/fpga-shells/xilinx/common/tcl/route.tcl**" and change to 
**phys_opt_design -directive AggressiveExplore**

Regarding the Boot : I personally suspect that the size of the linux image you are trying to boot is bigger then 30MB. 
If you want to stick to the "stock" boot loader (I do not), you can adjust the size by doing some math and change the line 12 in "**./fpga/src/main/resources/vcu118/sdboot/sd.c**" 
from **30<<20**  to your Linux image filesize (rounded to multiple of 512 =size of sector)

Charles-Tang, 2024-11-23:
> Hi, As a workaround for the timing problem. Revert back to 100Mhz it is fine but force the exploration of place & route:
>
> * Edit the line 11 in "**./fpga/fpga-shells/xilinx/common/tcl/route.tcl**" [...]

Hi,
thanks for reply, i tried the AggressiveExplore option, and it indeed improved the timing, but i still have the slow boot problem.
i've checked my linux image, and the size of br-base-bin-nodisk-flat is 24.8MiB, so this may not be the problem. 
Your idea inspired me to try fpga-shells overlay in earlier chipyard release, i will update once i get the results.
In addition, i noticed that in the L2 inclusiveCache Client Map there is a "debug" device, which did not show up in earlier versions, could this be the problem? And how can i remove this device?

L2 InclusiveCache Client Map:
        0 <= debug
        1 <= Core 0 DCache
        2 <= Core 0 ICache

Charles-Tang, 2024-11-25:
i tried replace the files in directory <chipyard>/fpga/src/main/resources/vcu118 with those from chipyard tag 1.8.1, and now it can boot normally. i suspect the frequency calculating in sd.c caused the problem, as changing march in Makefile should probably affect the behavior after the bbl. closing the issue since the problem is solved.

jerryz123, 2024-11-25:
Reopening this since this should not be an issue.

elialaz, 2025-01-01:
@Charles-Tang @jerryz123 I'm getting the same errors, but when I replace the files, it goes into a SD timeout Error. @Charles-Tang, have you changed other parameters?

Charles-Tang, 2025-04-15:
> [@Charles-Tang](https://github.com/Charles-Tang) [@jerryz123](https://github.com/jerryz123) I'm getting the same errors, but when I replace the files, it goes into a SD timeout Error. [@Charles-Tang](https://github.com/Charles-Tang), have you changed other parameters?
@elialaz ,i also see SD timeout errors sometimes, i solved it by replug the sd card, usually it works

elialaz, 2025-06-15:
@jerryz123 I found the error in the sd.c thanks to the suggestion of Amirmohammad Nazari in the google groups

`
diff --git a/fpga/src/main/resources/vcu118/sdboot/sd.c b/fpga/src/main/resources/vcu118/sdboot/sd.c
index bc1a68e8..162ec295 100644
--- a/fpga/src/main/resources/vcu118/sdboot/sd.c
+++ b/fpga/src/main/resources/vcu118/sdboot/sd.c
@@ -27,7 +27,7 @@
 // SPI SCLK frequency, in kHz
 // We are using the 25MHz High Speed mode. If this speed is not supported by the
 // SD card, consider changing to the Default Speed mode (12.5 MHz).
-#define SPI_CLK        25000
+#define SPI_CLK        12500
 
 // SPI clock divisor value
 // @see https://ucb-bar.gitbook.io/baremetal-ide/baremetal-ide/using-peripheral-devices/sifive-ips/serial-peripheral-interface-spi
@@ -90,7 +90,8 @@ static void sd_poweron(void)
        long i;
        // HACK: frequency change
 
-       REG32(spi, SPI_REG_SCKDIV) = SPI_DIV;
+       // REG32(spi, SPI_REG_SCKDIV) = SPI_DIV;
+       REG32(spi, SPI_REG_SCKDIV) = (F_CLK / 0.4);     // 400KHz for initialization
        REG32(spi, SPI_REG_CSMODE) = SPI_CSMODE_OFF;
        for (i = 10; i > 0; i--) {
                sd_dummy();
@@ -185,7 +186,8 @@ static int copy(void)
        kprintf("LOADING 0x%x B PAYLOAD\r\n", PAYLOAD_SIZE_B);
        kprintf("LOADING  ");
 
-       REG32(spi, SPI_REG_SCKDIV) = SPI_DIV;
+       // REG32(spi, SPI_REG_SCKDIV) = SPI_DIV;
+       REG32(spi, SPI_REG_SCKDIV) = (F_CLK / 5); // 5MHz
        if (sd_cmd(0x52, BBL_PARTITION_START_SECTOR, 0xE1) != 0x00) {
                sd_cmd_end();
                return 1;
`

this is the diff that Amirmohammad Nazari proposed and I verified works, in addition you need to remove the password from the buildroot-config line 9 of the base-br workload

- BR2_TARGET_GENERIC_ROOT_PASSWD="fpga"
+ BR2_TARGET_GENERIC_ROOT_PASSWD=""

If you apply these modification basically all issues regarding this matter can be closed