ucb-bar/chipyard #2126: Scala SBT Version Binary Compatibility Error

suhwanhong999, 2024-11-16:
### Background Work

- [X] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [X] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [X] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.11.0
Hash: ac58f38d77c99e9d1cafa64dfd6d4b00bdcd43e1

### OS Setup

Default Amazon F1 instance settings running on centos

### Other Setup

N/A

### Current Behavior

when running 'firesim buildbitstream', it runs into a version conflict error regarding sbt scala version compatibility. 

### Expected Behavior

launch build farm to return agfi to be added to the hwdb.yaml

### Other Information

[2024-11-16--02-17-39-buildbitstream-FWW710LQH4JX5TOS.log](https://github.com/user-attachments/files/17783244/2024-11-16--02-17-39-buildbitstream-FWW710LQH4JX5TOS.log)

suhwanhong999, 2024-11-20:
I have tried implementing the fix outlined in: https://stackoverflow.com/questions/74335368/scala-sbt-version-dependency-binary-compatibility-error-scala-xml,
but to no avail. Still having issues launching any workload using a gemmini accelerator.

jerryz123, 2024-11-25:
You may be able to find help faster by opening the issue in the firesim repo instead

closed 2024-12-07 by suhwanhong999