ucb-bar/chipyard #2235: Using Chipyard's default configuration ReRoCCManyGemminiConfig, the ReRoCC build failed.

silencerains, 2025-04-14:
### Background Work

- [x] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [x] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [x] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.13.0

### OS Setup

LSB Version:    core-11.1.0ubuntu2-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal
---
SHELL=/bin/bash
LANG=en_US.UTF-8
PATH=path
CONDA_PREFIX=/home/user/workspace/.conda-env
RISCV=/home/user/workspace]/.conda-env/riscv-tools
---
Python: 3.10.14
GCC: 13.2.0
CMAKE: 3.26.3
JAVA: OpenJDK 20.0.2
Verilator: 5.022

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

When building ReRoCC using the command: make run-binary-debug CONFIG=ReRoCCManyGemminiConfig BINARY=../../generators/rerocc/tests/rerocc.riscv -j 64
an error occurred. Additionally, I configured a custom SingleReRoCC configuration, but the same error persisted.
Technical details:
Command attempted: make run-binary-debug with ReRoCCManyGemminiConfig
Error message: Memory region 'ctrl' at List(AddressSet(0x20000, 0xfff)) only supports TransferSizes[1, 8] Get, but must support TransferSizes[1, 64]
Additional attempt: Custom SingleReRoCC configuration also failed with the same error
Issue: Transfer size mismatch in the control register region (supports 1-8 bytes but requires 1-64 bytes)
The specific error message is:
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at ... ()
        at chipyard.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:35)
        at chipyard.stage.phases.PreElaboration$$Lambda$498/485686996.apply(Unknown Source)
        at chisel3.Module$.evaluate(Module.scala:88)
        at chisel3.Module$.do_apply(Module.scala:35)
        at chisel3.stage.phases.Elaborate.$anonfun$transform$2(Elaborate.scala:52)
        at chisel3.stage.phases.Elaborate$$Lambda$512/405252209.apply(Unknown Source)
        at chisel3.internal.Builder$.$anonfun$buildImpl$1(Builder.scala:1042)
        at chisel3.internal.Builder$$$Lambda$515/1130551263.apply(Unknown Source)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
        at chisel3.internal.Builder$.buildImpl(Builder.scala:1032)
        at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:1023)
        at chisel3.internal.Builder$$$Lambda$513/1223050066.apply(Unknown Source)
        at logger.Logger$.$anonfun$makeScope$4(Logger.scala:148)
        at logger.Logger$$$Lambda$119/134310351.apply(Unknown Source)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
        at logger.Logger$.makeScope(Logger.scala:146)
        at logger.Logger$.makeScope(Logger.scala:133)
        at ... ()
        at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
Caused by: java.lang.IllegalArgumentException: requirement failed: Memory region 'ctrl' at List(AddressSet(0x20000, 0xfff)) only supports TransferSizes[1, 8] Get, but must support TransferSizes[1, 64]

### Expected Behavior

I have very limited knowledge about ReRoCC and I'm completely at a loss regarding this issue. Interestingly, when I used RocketGemminiConfig, no such error occurred.

jerryz123, 2025-04-16:
Could not reproduce. Please use latest main, not releases.

silencerains, 2025-04-21:
> Could not reproduce. Please use latest main, not releases.

silencerains, 2025-04-22:
> Could not reproduce. Please use latest main, not releases.
Jerry, thank you so much for your reply. I followed your suggestion and switched Chipyard to the main branch. However, I encountered some minor issues with the build environment. For instance, during step 3, I received the following error:
========== BEGINNING STEP 3: Building toolchain collateral ==========
Installing extra toolchain utilities/tests to /home/cp-stable/.conda-env/riscv-tools
==>  Installing Spike
=>  Starting riscv-isa-sim build
==>  Initializing riscv-isa-sim submodule
+ git submodule update --init --recursive /home/dys1/cp-stable/toolchains/riscv-tools/riscv-isa-sim
Cloning into '/home/dys1/cp-stable/toolchains/riscv-tools/riscv-isa-sim'...
Submodule path 'toolchains/riscv-tools/riscv-isa-sim': checked out 'b81b5a068636829eb7b299e63a32800b5dd047a0'
==>  cd /home/dys1/cp-stable/toolchains/riscv-tools/riscv-isa-sim
+ mkdir -p build
==>  Configuring riscv-isa-sim
+ cd build
+ ../configure --prefix=/home/cp-stable/.conda-env/riscv-tools --with-boost=no --with-boost-asio=no --with-boost-regex=no
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... configure: error: in `/home/cp-stable/toolchains/riscv-tools/riscv-isa-sim/build':
configure: error: cannot run C compiled programs.
If you meant to cross compile, use `--host'.
See `config.log' for more details
===================================
When I tried using the --host option, although it compiles, things didn't improve. During the simulator build, it appears that the system cannot find the corresponding glibc.
Could you suggest how I should address this problem?

adamsir32, 2025-04-24:
hello bro，i have the same question ，did you solve this problem？

suhwanhong999, 2025-06-17:
I am having the same issues as well. Using some of the latest commits to main, I am able to get past the setup step 3 issue for most of them but still run into the original problem when trying to make CONFIG=ReRoCCManyGemminiConfig. This issue doesn't happen for CONFIG=GemminiRocketConfig, for example.

suhwanhong999, 2025-06-17:
@jerryz123 As seen in https://github.com/ucb-bar/chipyard/issues/2241, I tried to use the latest version possible that would run on my stem (Linux), which was commit seen here (https://github.com/ucb-bar/chipyard/pull/2166). So, I am currently on b48e992, merged 1/10/25. While the build-setup.sh runs fine, I still run into the same issue originally described. Not sure why this issue persists