ucb-bar/chipyard #2241: chipyard 1.13 "riscv-isa-sim” build error!

zzulb, 2025-04-21:
### Background Work

- [x] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [x] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [x] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.13.0
latest

### OS Setup

ubuntu 20.0

### Other Setup

### Current Behavior

Execute “module_all riscv-isa-sim --prefix="${RISCV}" --with-boost=no --with-boost-asio=no --with-boost-regex=no” and an error is reported. It is prompted to give the --host parameter.

### Expected Behavior

"module_all riscv-isa-sim --prefix="${RISCV}" --with-boost=no --with-boost-asio=no --with-boost-regex=no --host=x86_64-linux-gnu" This can be executed. But I don't know if it is correct.

"module_all riscv-isa-sim --prefix="${RISCV}" --with-boost=no --with-boost-asio=no --with-boost-regex=no --host=riscv${XLEN}-unknown-elf" This still reports an error, missing libpthread, I don't know how to deal with it.

zzulb, 2025-04-22:
[image]

sdudyl, 2025-04-28:
可以交流一下？我也遇到了这个问题，大概知道原因，我的vx是13176659876

suhwanhong999, 2025-07-22:
doesn't seem to fix the issue as far as I can tell. I know the OS being old is the issue, since all of this works perfectly well on WSL (running on my laptop)