ucb-bar/chipyard #2256: java.lang.ClassNotFoundException: com.fasterxml.jackson.annotation.JsonView when running make for the default example in the simulator

ninomerlino, 2025-05-19:
### Background Work

- [x] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [x] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [x] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.12.0
Hash: 404c8d36

### OS Setup

Linux Host-002 6.14.4-300.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Apr 25 15:43:38 UTC 2025 x86_64 GNU/Linux
---
SYSTEMD_EXEC_PID=2258
MODULEPATH=/usr/share/Modules/modulefiles:/etc/modulefiles:/usr/share/modulefiles
SESSION_MANAGER=local/unix:@/tmp/.ICE-unix/2017,unix/unix:/tmp/.ICE-unix/2017
MOZ_GMP_PATH=/usr/lib64/mozilla/plugins/gmp-gmpopenh264/system-installed
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/0dd43446_0e45_4e10_ade0_680399fafbfb
MODULES_RUN_QUARANTINE=LD_LIBRARY_PATH LD_PRELOAD
LANG=en_US.UTF-8
MODULESHOME=/usr/share/Modules
WAYLAND_DISPLAY=wayland-0
XMODIFIERS=@im=ibus
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
LESSOPEN=||/usr/bin/lesspipe.sh %s
QT_IM_MODULE=ibus
USER=pinguinobiricchino
HOSTNAME=fedora
XDG_MENU_PREFIX=gnome-
HISTSIZE=50000
COLORTERM=truecolor
XDG_CURRENT_DESKTOP=GNOME
DESKTOP_SESSION=gnome
DOTNET_BUNDLE_EXTRACT_BASE_DIR=/home/pinguinobiricchino/.cache/dotnet_bundle_extract
OLDPWD=/home/pinguinobiricchino/Documenti/uni/12Chipyard/chipyard/sims
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
MANPATH=/usr/share/man:
PWD=/home/pinguinobiricchino/Documenti/uni/12Chipyard/chipyard
HOME=/home/pinguinobiricchino
MODULES_CMD=/usr/share/Modules/libexec/modulecmd.tcl
XDG_DATA_DIRS=/home/pinguinobiricchino/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share/:/usr/share/
VTE_VERSION=8001
XDG_SESSION_DESKTOP=gnome
DEBUGINFOD_IMA_CERT_PATH=/etc/keys/ima:
GNOME_SETUP_DISPLAY=:1
MAIL=/var/spool/mail/pinguinobiricchino
DEBUGINFOD_URLS=https://debuginfod.fedoraproject.org/ 
LOGNAME=pinguinobiricchino
GNOME_TERMINAL_SERVICE=:1.934
__MODULES_SHARE_MANPATH=:1
MEMORY_PRESSURE_WRITE=c29tZSAyMDAwMDAgMjAwMDAwMAA=
MEMORY_PRESSURE_WATCH=/sys/fs/cgroup/user.slice/user-1000.slice/user@1000.service/session.slice/org.gnome.SettingsDaemon.MediaKeys.service/memory.pressure
PATH=/home/pinguinobiricchino/Documenti/uni/12Chipyard/chipyard/software/firemarshal:/home/pinguinobiricchino/Documenti/uni/12Chipyard/chipyard/.conda-env/riscv-tools/bin:/home/pinguinobiricchino/Documenti/uni/12Chipyard/chipyard/.conda-env/bin:/home/pinguinobiricchino/miniforge3/condabin:/opt/sbt/bin:/usr/share/Modules/bin:/usr/lib64/ccache:/home/pinguinobiricchino/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/home/pinguinobiricchino/.dotnet/tools:/home/pinguinobiricchino/.cabal/bin:/home/pinguinobiricchino/.ghcup/bin:/usr/local/texlive/2024/bin/x86_64-linux:/home/pinguinobiricchino/miniforge3/bin
XDG_RUNTIME_DIR=/run/user/1000
DOTNET_ROOT=/usr/lib64/dotnet
SHELL=/bin/zsh
XDG_SESSION_TYPE=wayland
LOADEDMODULES=
SYSTEMD_SLEEP_FREEZE_USER_SESSIONS=0
GPG_TTY=/dev/pts/0
EDITOR=/usr/bin/nano
GDM_LANG=en_US.UTF-8
XAUTHORITY=/run/user/1000/.mutter-Xwaylandauth.EGMJ62
HISTCONTROL=ignoredups
__MODULES_LMINIT=module use --append /usr/share/Modules/modulefiles:module use --append /etc/modulefiles:module use --append /usr/share/modulefiles
DISPLAY=:0
TERM=xterm-256color
USERNAME=pinguinobiricchino
[... 631 lines cut ...]
	at firrtl.options.Stage$$anon$1.transform(Stage.scala:42)
	at firrtl.options.Stage$$anon$1.transform(Stage.scala:42)
	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:45)
	at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:183)
	at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:179)
	at scala.collection.immutable.List.foldLeft(List.scala:79)
	at firrtl.options.Stage.$anonfun$transform$2(Stage.scala:45)
	at logger.Logger$.$anonfun$makeScope$4(Logger.scala:148)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
	at logger.Logger$.makeScope(Logger.scala:146)
	at logger.Logger$.makeScope(Logger.scala:125)
	at firrtl.options.Stage.transform(Stage.scala:45)
	at firrtl.options.Stage.execute(Stage.scala:56)
	at firrtl.options.StageMain.main(Stage.scala:69)
	at chipyard.Generator.main(Generator.scala)
Caused by: java.lang.ClassNotFoundException: com.fasterxml.jackson.annotation.JsonView
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:641)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:188)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	... 165 more
```

### Expected Behavior

Doesn't compile the simulator

berkaybuyukdere, 2025-05-19:
**- java.lang.ClassNotFoundException: com.fasterxml.jackson.annotation.JsonView**

Rebuild the toolchains

`cd chipyard
./scripts/build-toolchains.sh`

or you can manually add the missing jackson dependency 

```
cd ~/.ivy2/jars
wget https://repo1.maven.org/maven2/com/fasterxml/jackson/core/jackson-annotations/2.14.2/jackson-annotations-2.14.2.jar
```

for the last part:

clean and rebuild.

```
cd chipyard/sims/verilator
rm -rf target/
make clean
make
```

ninomerlino, 2025-05-19:
There is no script `build-toolchains.sh` and i don't have the directory `~/.ivy2/jars`

berkaybuyukdere, 2025-05-19:
> There is no script `build-toolchains.sh` and i don't have the directory `~/.ivy2/jars`

sorry for the previous misunderstanding.
try this SBT direct method:

`cd chipyard/sims/verilator
sbt "set libraryDependencies += \"com.fasterxml.jackson.core\" % \"jackson-annotations\" % \"2.14.2\""
sbt "set libraryDependencies += \"com.fasterxml.jackson.core\" % \"jackson-core\" % \"2.14.2\""
sbt "set libraryDependencies += \"com.fasterxml.jackson.core\" % \"jackson-databind\" % \"2.14.2\""`

after applying this:

cd chipyard/sims/verilator
make clean
make

ninomerlino, 2025-05-19:
doesn't seem to work, I still get the same error.

ninomerlino, 2025-05-20:
Adding
```
"com.fasterxml.jackson.core" % "jackson-annotations" % "2.13.5",
"com.fasterxml.jackson.core" % "jackson-databind" % "2.13.5",
```
to
```
lazy val rocketchip = freshProject("rocketchip", rocketChipDir)
  .dependsOn(hardfloat, rocketMacros, diplomacy, cde)
  .settings(commonSettings)
  .settings(chiselSettings)
  .settings(
    libraryDependencies ++= Seq(
      "com.lihaoyi" %% "mainargs" % "0.5.0",
      "com.fasterxml.jackson.core" % "jackson-annotations" % "2.13.5",
      "com.fasterxml.jackson.core" % "jackson-databind" % "2.13.5",
      "org.json4s" %% "json4s-jackson" % "4.0.5",
      "org.scalatest" %% "scalatest" % "3.2.0" % "test",
      "org.scala-graph" %% "graph-core" % "1.13.5"
    )
  )
```
in `build.sbt` seems to have solved the issues

berkaybuyukdere, 2025-05-20:
ahh

`cd chipyard
`

```
libraryDependencies ++= Seq(
  "com.fasterxml.jackson.core" % "jackson-annotations" % "2.14.2",
  "com.fasterxml.jackson.core" % "jackson-core" % "2.14.2",
  "com.fasterxml.jackson.core" % "jackson-databind" % "2.14.2"
)
```

I did a similar experiment and I used this method 

It worked when I did it directly in build.sbt.