ucb-bar/chipyard #2270: Error while running ./build-setup.sh riscv-tools -v

Rocketuser01, 2025-07-22:
### Background Work

- [x] Yes, I searched the [mailing list](https://groups.google.com/forum/#!forum/chipyard)
- [x] Yes, I searched [prior issues](https://github.com/ucb-bar/chipyard/issues)
- [x] Yes, I searched the [documentation](https://chipyard.readthedocs.io/)

### Chipyard Version and Hash

Release: 1.5.0
Hash: a6a6a6
Hi, I'm encountering some problems when running the build script, help would be much appreciated!

Conda version: 25.5.1
python version 3.13.5

How to reproduce:
run: 

git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
git checkout HEAD
./build-setup.sh riscv-tools

```
./build-setup.sh riscv-tools -v
+ shift
+ '[' '' '!=' '' ']'
+ tee build-setup.log
+ set +e
+ replace_content env.sh cy-dir-helper CY_DIR=/home/user/chipyard
++ uname -s
+ '[' Linux = Darwin ']'
+ READLINK=readlink
+++ readlink -f /home/user/chipyard/scripts/utils.sh
++ dirname /home/user/chipyard/scripts/utils.sh
+ DIR=/home/user/chipyard/scripts
+ file=env.sh
+ shift
+ key=cy-dir-helper
+ shift
+ /home/user/chipyard/scripts/replace-content.py env.sh cy-dir-helper CY_DIR=/home/user/chipyard
+ run_step 1
+ local value=1
+ [[ !    =~  1  ]]
+ begin_step 1 'Conda environment setup'
+ thisStepNum=1
+ thisStepDesc='Conda environment setup'
+ echo ' ========== BEGINNING STEP 1: Conda environment setup =========='
 ========== BEGINNING STEP 1: Conda environment setup ==========
+ CONDA_REQS=/home/user/chipyard/conda-reqs
+ CONDA_LOCK_REQS=/home/user/chipyard/conda-reqs/conda-lock-reqs
+ '[' false = false ']'
+ LOCKFILE=/home/user/chipyard/conda-reqs/conda-lock-reqs/conda-requirements-riscv-tools-linux-64.conda-lock.yml
+ CONDA_LOCK_ENV_PATH=/home/user/chipyard/.conda-lock-env
+ '[' -d /home/user/chipyard/.conda-lock-env ']'
+ '[' -d /home/user/chipyard/.conda-env ']'
+ rm -rf /home/user/chipyard/.conda-lock-env
++ grep conda-lock /home/user/chipyard/conda-reqs/chipyard-base.yaml
++ sed 's/^ \+-//'
+ conda create -y -p /home/user/chipyard/.conda-lock-env -c conda-forge conda-lock=2.5.7
2 channel Terms of Service accepted
Channels:
 - conda-forge
 - defaults
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

[... 486 lines cut ...]
LC_ALL=C
CONDA_BACKUP_PATH=/home/user/chipyard/.conda-env/riscv-tools/bin:/home/user/chipyard/software/firemarshal:/home/user/chipyard/.conda-env/riscv-tools/bin:/home/user/.pyenv/shims:/home/user/.pyenv/bin:/home/user/anaconda3/condabin:/home/user/.local/bin:/home/user/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin
XDG_DATA_DIRS=/home/user/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share:/usr/share
PATH=/home/user/chipyard/.conda-env/bin:/home/user/.pyenv/shims:/home/user/.pyenv/bin:/home/user/anaconda3/condabin:/home/user/.local/bin:/home/user/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbinconda/bin
SELINUX_LEVEL_REQUESTED=
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
MAIL=/var/spool/mail/user
SSH_TTY=/dev/pts/0
OLDPWD=/home/user
BASH_FUNC_which%%=() {  ( alias;
 eval ${which_declare} ) | /usr/bin/which --tty-only --read-alias --read-functions --show-tilde --show-dot $@
}
_=/usr/bin/printenv

### Other Setup

Ex: Prior steps taken / Documentation Followed / etc...

### Current Behavior

Expecting problems when running the build script

### Expected Behavior

Build script runs without error

suhwanhong999, 2025-07-22:
What is the exact error you are getting? It also says you are using "Release: 1.5.0" but you also run 'git checkout HEAD', which leads me to assume you are running the latest commit (which is always recommended).

Rocketuser01, 2025-07-29:
Alright, so letting the script print out INFO:root: for 5 (!) minutes causes it to then continue running normally. The specific error I get after that is 

`gmake:` Leaving directory `/home/user/chipyard/toolchains/riscv-tools/riscv-tests/build'
==> Installing espresso logic minimizer
/home/user/chipyard/scripts/build-toolchain-extra.sh: line 96: scripts/install-espresso.sh: No such file or directory
build-setup.sh: Build script failed with exit code 127 at step 3: Building toolchain collateral
`

Rocketuser01, 2025-08-12:
After uninstalling conda as per instructions and reinstalling using 
curl -O https://repo.anaconda.com/archive/Anaconda3-2025.06-0-Linux-x86_64.sh
following this link 
https://www.anaconda.com/docs/getting-started/anaconda/install#macos-linux-installation:how-do-i-verify-my-installers-integrity
the build now worked and I will close the issue.
@suhwanhong999 I have the suspicion that the weird INFO:root: error stems from a rendering issue. Because in this build at first it spammed this message, then suddenly text popped up behind the INFO:root: even though it was not there before. Just so you know if someone else mentions that. Many thanks!

closed 2025-08-12 by Rocketuser01